nasa/fprime #2074: Hardcoded buffer size in TcpClient/TcpDriver

thomas-bc, 2023-06-08:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

See https://github.com/nasa/fprime/blob/da4b1daae77f46a940733fbb20a4be8efd3a89b8/Drv/TcpClient/TcpClientComponentImpl.cpp#L50-L51

We should do like UARTDriver (see [here](https://github.com/nasa/fprime/blob/da4b1daae77f46a940733fbb20a4be8efd3a89b8/Drv/LinuxUartDriver/LinuxUartDriver.cpp#L46-L47)) and configure the size at configure() time so that it can be configured for each instance in the topology. Set a default size of 1024 so that it doesn't break any existing code.

LeStarch, 2024-06-07:
@menaman123 the fix would be to add an "allocation size" to the `configure` method:

https://github.com/nasa/fprime/blob/da4b1daae77f46a940733fbb20a4be8efd3a89b8/Drv/TcpClient/TcpClientComponentImpl.cpp#L32-L36

bdshenker, 2024-10-28:
This issue appears to have been fixed in commit 7FF002C by menamen123:

```
TcpClientComponentImpl.hpp
  PROTECTED: FwSizeType m_allocation_size;
  SocketIpStatus configure(const char* hostname,
                             const U16 port,
                             const U32 send_timeout_seconds = SOCKET_SEND_TIMEOUT_SECONDS,
                             const U32 send_timeout_microseconds = SOCKET_SEND_TIMEOUT_MICROSECONDS,
                             FwSizeType buffer_size = 1024);
TcpClientComponentImpl.cpp
  Fw::Buffer TcpClientComponentImpl::getBuffer() {
      return allocate_out(0, static_cast<U32>(m_allocation_size));
  }
```
If I am correct then this ticket can now be closed per GIT ticket # 2759.

LeStarch, 2024-10-29:
@bdshenker the above ticket did resolve the problem in TcpClient, but not TcpServer nor Udp.

https://github.com/nasa/fprime/blob/0e4b4c2aa219ed23ac7fe03b6b4196146ae7e8e3/Drv/TcpServer/TcpServerComponentImpl.cpp#L50

bdshenker, 2024-10-29:
@LeStarch I completed Git#2074 and reran regression suite locally.
However, when making pull request, one of the integration tests failed. 
Please advise how I should proceed.

LeStarch, 2024-11-22:
Closing as complete.

closed 2024-11-22 by LeStarch