nasa/fprime #2075: Switch Framer and Deframer to use Mallocator Pattern

LeStarch, 2023-06-08:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

Mallocator pattern is preferred over member-allocated buffers.

matt392code, 2025-01-08:
Proposed specialized allocator for F' that would be suitable for spacecraft software:
[framer-example.txt](https://github.com/user-attachments/files/18342580/framer-example.txt)
Key features of this implementation:

1. Memory Safety:
- Explicit tracking of memory allocations
- Checks against maximum spacecraft heap size
- Exception handling for memory allocation failures
- Memory usage statistics for monitoring

2. Spacecraft-Specific Features:
- Memory usage tracking (total, peak, current)
- Configurable maximum heap size
- All operations are noexcept where possible
- Namespace integration with F' framework (Fw)

3. Integration Example for Framer:
Here's how you would modify a Framer component to use this allocator:
[framer-example.txt](https://github.com/user-attachments/files/18342580/framer-example.txt)

4. Benefits for F' Framework:
- Better memory tracking for spacecraft operations
- Consistent allocation strategy across components
- Memory usage visibility for debugging and monitoring
- Safe handling of spacecraft memory constraints

LeStarch, 2025-01-23:
@t@thomas-bc this likely now applies to the accumulator.

closed 2025-04-11 by LeStarch