nasa/fprime #2100: Add Table of Contents to side bar in F´ docs website

thomas-bc, 2023-06-23:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**| Docs |
---
## Feature Description

Here's our Jekyll theme: https://github.com/fprime-community/fprime-theme. We are using a clean slate theme that displays a single page at a time.
From the popular Jekyll themes, it looks like there's a bunch of them that have a Table of Contents in a sidebar https://jekyllthemes.io/jekyll-documentation-themes 
It would be quite nice if we could have that as well. Not sure how easy/doable it would be to do with our current theme. Posting this issue here in case anyone want to take a bite at doing that!

## Rationale

Improve navigability of F´ docs

Chr1st1anSears, 2023-08-15:
Hey @thomas-bc :wave: how do you like the [Just the Docs theme](https://just-the-docs.github.io/just-the-docs/), or [the Jekyll Doc theme](https://jekyllthemes.io/theme/documentation)? I prefer Just the Docs, but it's your site!

I'm a technical writer for Red Hat, and as a big fan of NASA,  I've wanted to make an open source contribution to your org for a while. If you all are still interested in making this upgrade to the F' docs set, I'd be happy to give it a try!

LeStarch, 2023-08-16:
Our goal is to keep our branding which consists of two elements:

1. Our banner image
2. Our color pallet

To accomplish this, we took the slate theme and modified it to use those two elements.   If we could use the same process to adapt the Just the Docs or Jekyll themes, that would be awesome.

Here is our theme repository for reference:
https://github.com/fprime-community/fprime-theme

LeStarch, 2023-08-16:
Here are the color definitions:L

https://github.com/fprime-community/fprime-theme/blob/a6a3660c7734a49efefe1565dd2c03f98f16ba62/_sass/fprime-theme.scss#L6-L7C27

Chr1st1anSears, 2023-08-16:
Thanks @LeStarch! That banner image with Ingenuity is awesome, and I figured you'd want to keep that, along with the color scheme. 

I'll be on vacation for the rest of this week, but next week I'll try some in-depth tinkering with the themes and let you know how it goes.

Chr1st1anSears, 2023-08-25:
Hey @LeStarch and @thomas-bc :wave: 

Here's what I've got so far: https://chr1st1ansears.github.io/fprime/

What do you think? I don't think it's finished yet. And there are lots of small technical decisions/questions I could go into (example: for a number of reasons, this doesn't use a new Jekyll theme, it builds on your existing theme). But I'd love to hear any feedback you have at this point. Thanks.

thomas-bc, 2023-08-25:
This is really cool! Will help a lot with navigability, thanks!
A couple notes:
- We probably should include in the ToC one level up of the hierarchy, with basically [these top level sections](https://nasa.github.io/fprime/latest), and include the current contents under the "User Guide" section. With this, it should be possible to make the top level sections clickable links as well? 
- With a narrow browser window (or when using a phone), the contents of the side bar get superposed to the main content, which we will want to avoid. Maybe hide the content or the ToC when the window is too narrow?
- Curious: what are the reasons you chose not to build on a new Jekyll theme?

Chr1st1anSears, 2023-08-25:
Hey Thomas--thanks for your feedback! I'm glad you think it'll help. Concerning your notes:
- We can definitely make top-level sections clickable links, too. And do I understand correctly that you'd want it to be structured like this?
**********************
Installation
Tutorials 
              HelloWorld
              LedBlinker 
               ...
User Guide
             Getting Started with F'
                            What is F': a brief introduction
                            Installing F'
                            ...
             F' Users Manual: an in-depth description of F' concepts
                            A More Complete Introduction to F'
                            ...
            ...
Design and Philosophy
**********************
- I agree, the overlapping text is definitely an issue. I haven't figured out how to fix it yet, but I won't make a PR until I do.
- I haven't totally abandoned the ideas of themes yet, but I think it would be quite challenging. With the Just The Docs theme, you'd have to adjust the YAML front matter of _every individual_ doc, in order to determine its place in the nav bar. The other theme might work, but I wasn't sure how to get the branding right. 

Please let me know if I understand your goal for the hierarchy, or if you have any other feedback. Also BTW, I won't be able to work on this much if at all next week, but the week after, I'll be back at it.

Have a great weekend.

Chr1st1anSears, 2023-09-05:
@thomas-bc would you please take a look at my fork's updated site? https://chr1st1ansears.github.io/fprime/

Is that what you had in mind with the hierarchy of the actual content of the nav bar? You'll also notice that the side bar and main content no longer overlap if you condense the screen. It looks especially improved, I think, on a phone's browser.

There's definitely still some fine-tuning to do. For one thing, the spacing on the page isn't quite right; the nav bar takes up more blank space around it than it should. Still, at this point, what you see on my fork's site is pretty close to the functionality I'd try to add with a PR. Please let me know your thoughts. Thanks!

thomas-bc, 2023-09-05:
This is what I had in mind yes. This is awesome!

Agreed that it needs some aesthetic/usability fine-tuning but this is really useful already. Feel free to open the PR anytime, we can also help to take this across the finish line if you don't have the time to do so.

Thanks again!

Chr1st1anSears, 2023-09-06:
Thanks @thomas-bc! I'll open up a PR this week or next :+1:

Chr1st1anSears, 2023-09-08:
Hey @thomas-bc, I'm sure you'll already be notified, but I opened several PRs to add the nav bar to your docs site. Most of the proposed changes are actually to the fprime-theme repo:

- https://github.com/fprime-community/fprime-theme/pull/2

- https://github.com/fprime-community/fprime-theme/pull/3

- https://github.com/nasa/fprime/pull/2271

Over the last few days, I worked on improving the aesthetics of the nav. My wife thought it now looks "sleek," I hope you might agree :)

Regardless, please let me know if there is anything else I can do or questions about my code I can answer to help get this thing over the finish line. And thanks for working with me. It was effort-intensive--I'd never done nearly so much with HTML/CSS before--but it was also an honor to collaborate with you on something that might help us, in some small way, get out there :rocket:

thomas-bc, 2023-09-09:
@Chr1st1anSears Thank you very much for all of this! Fair warning this may take a little bit to review: this is a pretty substantial change, and we also have other things going on right now. We'll get back to you ASAP!

Chr1st1anSears, 2023-09-11:
@thomas-bc understood, no rush :)

thomas-bc, 2023-10-03:
https://nasa.github.io/fprime/ 🔥 🔥 🔥 

This is awesome! I'll try to get TOCs in the tutorial repositories as well, so the navigation can be seamless between the main site and the Tutorial links. 

Quick question @Chr1st1anSears, do you know if there's a way to control from GitHub where the site is looking for the `_includes/toc.md` file? Or another solution to the following issue:
I tried adding a toc.md [here](https://github.com/fprime-community/fprime-workshop-led-blinker/compare/main...thomas-bc:fprime-workshop-led-blinker:main), but I'm getting errors during build saying it can't find `_includes/toc.md`. I think this is because in the core nasa/fprime repo, we build the docs from the `docs/` folder (see screenshot below), whereas we build from top-level of the tutorial repositories (the objective is to get the root README.md as the landing page for the website). 

[image]

A solution may be to simply add the TOC in `/_includes/toc.md` rather than `/docs/_includes/toc.md` on the tutorial repos, but that would be a little odd. Trying to keep everything under /docs here.

Chr1st1anSears, 2023-10-03:
@thomas-bc thank you so much!! It is truly an honor and a privilege to collaborate with you and your team on this.

Concerning the snag around _includes, I did a little research, and it looks like you can override where GH looks for that folder. Specifically, you can add the following line to _config.yml:
`includes_dir: docs/_includes`
And I believe you'll want to add that line to _config.yml in the tutorial repos, not fprime-theme. 

One other thing--I actually applied for a job opening as a technical writer at JPL. The job req is R4664, I submitted my application on 9/13, and my name (as is probably obvious from my GH username) is Christian Sears. If you're willing and able, would you please put in a good word with HR for me? If for any reason you're not--no worries! But I'd be deeply grateful for it if you could pass along any recommendation.

Regardless, please let me know if that idea for the _includes directory helps, and if there's anything else I can do to help!

thomas-bc, 2023-10-05:
I was able to use that `include: ` field in the tutorial, awesome, thank you very much!

There's not much we can do about job postings unfortunately. Do mention your contribution to F' if/when you talk to someone though :) I'll remember that for sure. Best of luck!

Chr1st1anSears, 2023-10-05:
Glad to hear that about the `include: ` field. And thanks! I'll definitely bring up the contribution if I get a chance.

thomas-bc, 2023-10-10:
I'll take care of this unless someone else wants to, but here's the list of tutorials we need to add toc.md to:
- [x] LedBlinker - https://github.com/fprime-community/fprime-workshop-led-blinker/pull/45
- [ ] https://github.com/fprime-community/fprime-tutorial-hello-world
- [ ] https://github.com/fprime-community/fprime-tutorial-math-component
- [ ] https://github.com/fprime-community/fprime-tutorial-arduino-blinker

Chr1st1anSears, 2023-10-11:
I think tomorrow I should have time to do at least one of those :+1: I'll start with arduino blinker to avoid repeating work

Chr1st1anSears, 2023-10-12:
I probably won't have time for the other ones, but I did put up a PR for arduino blinker: https://github.com/fprime-community/fprime-tutorial-arduino-blinker/pull/6

thomas-bc, 2023-10-23:
All done! Thank you so much once again @Chr1st1anSears 🚀 
This is looking great

Closing this issue as completed 😄

closed 2023-10-23 by thomas-bc

Chr1st1anSears, 2023-10-23:
@thomas-bc my pleasure--cheers :clinking_glasses: