nasa/fprime #2107: Add visualization to math component tutorial

hAkselS, 2023-06-28:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

The math component tutorial completely leaves out all mentions of the F' visualizer. Please add the visualized to the math component tutorial.

thomas-bc, 2023-07-11:
Waiting on next release which should include https://github.com/nasa/fprime/issues/2059

thomas-bc, 2024-03-22:
See https://github.com/fprime-community/fprime-tools/pull/147

The necessary steps are
- fprime-util generate
- fprime-util build
- cd MyDeployment/Top
- fprime-util visualize

To be added to https://github.com/fprime-community/fprime-tutorial-math-component

If anyone would like to contribute, would be amazing to have this!

matt392code, 2025-01-09:
Proposed visualization steps for the F' MathComponent tutorial:
[math-component-viz.md](https://github.com/user-attachments/files/18357643/math-component-viz.md)

The content follows the tutorial's existing style and includes:
1. Step-by-step instructions for using the visualizer
2. Explanation of what the visualization shows
3. How to interpret and interact with the visualization
4. Debugging tips and common issues
5. Next steps and practical applications

To integrate this into the tutorial:
1. Add it as a new section after "8. Error handling"
2. Update the table of contents to include "9. Visualization"
3. Add any necessary cross-references from other sections

matt392code, 2025-01-09:
[image]

matt392code, 2025-01-09:
[math-component-viz(2).md](https://github.com/user-attachments/files/18357733/math-component-viz.2.md)
The visualizations use standard F' color coding:
- Green for input ports
- Blue for output ports
- Yellow for command ports
- Purple for event ports
- Orange for telemetry ports
[image]

closed 2025-03-31 by thomas-bc