nasa/fprime #2108: Baremetal TaskRunner Priority

ethancheez, 2023-06-28:
Implement priorities for the baremetal taskrunner. Higher priority tasks will be run first.

matt392code, 2025-01-09:
Proposed implementation of a priority-based TaskRunner for F Prime's baremetal systems:
[task-runner.txt](https://github.com/user-attachments/files/18357891/task-runner.txt)
 Here are the key features and design decisions:

1. Priority System
   - Tasks are assigned a priority level (U32)
   - Higher numbers indicate higher priority
   - Default priority is 0 if not specified

2. Task Management
   - Tasks are stored in a sorted vector of PriorityTask structures
   - The vector is maintained in priority order (highest to lowest)
   - Sorting happens on task addition

3. Key Methods
   - `addTask(Task*, U32)`: Adds a task with specified priority
   - `removeTask(Task*)`: Removes a task from the registry
   - `run()`: Executes tasks in priority order
   - `stop()`: Safely stops task execution

4. Thread Safety Considerations
   - The implementation avoids using U32 for thread synchronization
   - Task addition/removal is safe during non-execution phases

ethancheez, 2025-01-09:
I believe this was added in `fprime-baremetal`. You can find the change in [this commit](https://github.com/fprime-community/fprime-baremetal/commit/6fa7db996632d9cd31bd87137f98e68ab2f1b939)

closed 2025-01-09 by ethancheez