nasa/fprime #2109: microFS Implementation on Baremetal

ethancheez, 2023-06-28:
Implement microFS into the Baremetal Reference

LeStarch, 2024-03-27:
@timcanham you agreed to take on this issue, although coordination with @rlcheng might help as this is a bit of an overlap.

matt392code, 2025-01-08:
Proposed steps:
First, let's create a baremetal-compatible file system interface that follows microFS patterns:
[baremetal-fs.txt](https://github.com/user-attachments/files/18342758/baremetal-fs.txt)

Now let's implement the hardware abstraction layer for storage access:
[storage-hal.txt](https://github.com/user-attachments/files/18342763/storage-hal.txt)

Key aspects of this implementation:
1. File System Layer:
- Implements core microFS operations (ls, rm, put, get)
- Adds baremetal-specific features (open, close, format)
- Fixed-size file table for embedded use
- Simple but robust file allocation system
- Built-in defragmentation support

2. Hardware Abstraction Layer:
- Direct hardware register access for storage control
- Block-aligned operations for flash memory
- Error checking and status monitoring
- Separate read/write control
- Hardware initialization support

3. Key Features:
- No dynamic memory allocation
- Fixed buffer sizes suitable for embedded systems
- Error handling appropriate for baremetal
- Direct hardware access
- Block-aligned operations

4. Integration Notes:
- The implementation needs to be initialized with a storage base address
- File operations are synchronous
- All operations check for hardware errors
- Block alignment is enforced for flash memory compatibility

matt392code, 2025-01-08:
SVG diagram of implementation:
[image]

kevin-f-ortega, 2025-01-28:
I plan to put microFS into [`fprime-baremetal`](https://github.com/fprime-community/fprime-baremetal).

Also, I will be using the latest F Prime devel since it has OSAL and quality-of-life updates (e.g., `register_os_implementation`); so, I'll be doing some minor updates to OSAL. FYI @ethancheez

thomas-bc, 2025-04-11:
Slipping to next sprint: developer unavailability

kevin-f-ortega, 2025-04-30:
Addressed here https://github.com/fprime-community/fprime-baremetal/pull/4