nasa/fprime #2110: Spell check math component tutorial

hAkselS, 2023-06-28:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

Math component tutorial was written and pushed without any spell check.

thomas-bc, 2023-07-11:
- [ ] https://github.com/fprime-community/fprime-tutorial-hello-world
- [ ] https://github.com/fprime-community/fprime-tutorial-math-component
- [ ] https://github.com/fprime-community/fprime-workshop-led-blinker

matt392code, 2025-01-09:
Proposed GitHub Actions workflow for spell checking that can be used across all three repositories:
[spellcheck-workflow.txt](https://github.com/user-attachments/files/18357975/spellcheck-workflow.txt)
Here are the key features:

1. Triggers on:
   - Push to main and devel branches
   - Pull requests to main and devel
   - Manual workflow dispatch

2. Uses pyspelling with hunspell for spell checking

3. Includes custom configuration:
   - Checks all Markdown files
   - Excludes the fprime submodule directory
   - Ignores code blocks and HTML comments
   - Custom dictionary for F´ specific terms

4. Provides detailed error reporting
To implement this in each repository:
1. Create `.github/workflows/spellcheck.yml` in each repository and copy the workflow content
2. Both `.spellcheck.yaml` and `.wordlist.txt` will be created automatically by the workflow
3. The workflow will run automatically on PRs and pushes