nasa/fprime #2112: Add code coverage link

hAkselS, 2023-06-28:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

So far, no tutorial covers checking unit test code coverage in detail. A github markdown should be created that explains how to check for code coverage. This markdown can be linked in any tutorial that mentions unit testing.

matt392code, 2024-12-27:
[code-coverage-tutorial.md](https://github.com/user-attachments/files/18258474/code-coverage-tutorial.md)
The tutorial covers:
 - Setting up the necessary tools
 - Configuring F Prime projects for coverage analysis
 - Running tests with coverage
 - Understanding coverage reports
 - Best practices and common issues
 - CI/CD integration
 - Troubleshooting guide
The tutorial can be linked from other F Prime documentation that mentions unit testing.

thomas-bc, 2025-03-28:
```
fprime-util generate --ut
fprime-util build --ut 
fprime-util check --coverage
```

This will generate a coverage/ directory that has HTML files which shows code coverage info

closed 2025-03-31 by thomas-bc