nasa/fprime #2116: Create RPi PWM driver

hAkselS, 2023-06-28:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Create a driver to allow users to control the raspberry pi's PWM pins. Preferably, this is done using hardware PWM, not software PWM.

## Rationale
This feature is needed for the F' system reference's use of a rotor.

matt392code, 2025-01-09:
Proposed PWM driver:
[pwm-driver.txt](https://github.com/user-attachments/files/18358339/pwm-driver.txt)
This PWM driver component for F Prime implements the following key features:
1. Hardware PWM support for Raspberry Pi GPIO pins 18 and 19 (which support hardware PWM)
2. Frequency range from 10 Hz to 8 KHz
3. Duty cycle control from 0-100%
4. Command handlers for:
   - Setting PWM frequency 
   - Setting duty cycle
   - Starting PWM output
   - Stopping PWM output
5. Input validation and error handling
6. Uses the pigpio library for low-level hardware control

Also note that you'll need to:
1. Install the pigpio library on the Raspberry Pi
2. Link against libpigpio in your CMake configuration
3. Run the pigpio daemon (pigpiod) before using hardware PWM

matt392code, 2025-01-09:
Proposed FPP model file for the PWM driver component:
[pwm-fpp.txt](https://github.com/user-attachments/files/18358359/pwm-fpp.txt)
This FPP model file defines:
1. Commands:
   - PWM_SET_FREQUENCY: Set the PWM frequency (10-8000 Hz)
   - PWM_SET_DUTY_CYCLE: Set the duty cycle (0-100%)
   - PWM_START: Start PWM output on a pin
   - PWM_STOP: Stop PWM output on a pin

2. Events:
   - PWM_INIT_ERROR: PWM initialization failed
   - PWM_INVALID_PIN: Invalid pin specified
   - PWM_INVALID_FREQUENCY: Invalid frequency specified 
   - PWM_INVALID_DUTY_CYCLE: Invalid duty cycle specified
   - PWM_HARDWARE_ERROR: Hardware error occurred

3. Telemetry:
   - PWM_FREQUENCY: Current PWM frequency
   - PWM_DUTY_CYCLE: Current PWM duty cycle
   - PWM_ACTIVE: PWM active status

4. Required ports for F Prime framework integration:
   - Time get port
   - Command ports (receive, registration, response)
   - Event port
   - Telemetry port
   - Text event port

To build the component, you'll need to:
1. Place both files in your F Prime project structure:
   ```
   YourProject/
     Drv/
       PwmDriver/
         PwmDriverComponentImpl.hpp
         PwmDriverComponentImpl.cpp
         PwmDriver.fpp
   ```
2. Add a CMakeLists.txt file to integrate with the build system
3. Update your topology file to include this component

matt392code, 2025-01-09:
Proposed CMakeLists.txt file to integrate the PWM driver into the F Prime build system:
[pwm-cmake.txt](https://github.com/user-attachments/files/18358378/pwm-cmake.txt)
This CMakeLists.txt file sets up the build configuration for the PWM driver component. Key features include:
1. Specifies source files for compilation
2. Generates F Prime component base files from the FPP model
3. Registers the component with F Prime's build system
4. Links required libraries:
   - F Prime framework components (Fw_Comp, Fw_Cmd, etc.)
   - pigpio library for Raspberry Pi GPIO control
5. Adds necessary compiler definitions and include directories

To complete the component integration, you'll need to:
1. Update your project's main CMakeLists.txt to include this component's directory:
```cmake
add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/Drv/PwmDriver/")
```

2. Install the pigpio library on your system:
```bash
sudo apt-get update
sudo apt-get install pigpio
```

3. Make sure the pigpio daemon is running:
```bash
sudo systemctl enable pigpiod
sudo systemctl start pigpiod
```
4. Update your topology file to instantiate and connect the component.

matt392code, 2025-01-09:
Proposed example topology configuration to show how to integrate the PWM driver component:
[pwm-topology.txt](https://github.com/user-attachments/files/18358408/pwm-topology.txt)
This topology example shows how to integrate the PWM driver component into an F Prime system. 
Key features include:
1. Component Instantiation:
   - Creates PWM driver instance with base ID 0x0E00
   - Sets up queue size, stack size, and priority

2. Standard F Prime Component Integration:
   - Rate Group for scheduling
   - Command Dispatcher for command handling
   - System Time for timestamps
   - Event Logger for event handling
   - Telemetry Channel for telemetry
   - Text Logger for text events

3. Port Connections:
   - Command registration and handling
   - Telemetry output
   - Event logging
   - Time services
   - Text event logging

To use this topology:
1. Save it as `Example.fpp` in your project's topology directory
2. Create a deployment CMakeLists.txt that includes:
```cmake
add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/PwmExample/")
```
3. Generate the topology:
```bash
fprime-util generate
```
4. Build the deployment:
```bash
fprime-util build
```
You can then interact with the PWM driver through the F Prime ground system using commands like:
```python
# Start PWM on GPIO 18
fprime-cli command-send pwmDriver.PWM_START 18

# Set frequency to 1000 Hz
fprime-cli command-send pwmDriver.PWM_SET_FREQUENCY 18 1000

# Set duty cycle to 50%
fprime-cli command-send pwmDriver.PWM_SET_DUTY_CYCLE 18 50.0

# Stop PWM
fprime-cli command-send pwmDriver.PWM_STOP 18
```