nasa/fprime #2117: PWM manager

hAkselS, 2023-06-28:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Add a PWM manager for the specific rotor that we are using in the system reference. The manager should allow the rotor control application to pass angle arguments.

thomas-bc, 2023-07-26:
https://github.com/hAkselS/fprime-system-reference/tree/dev/actuator

thomas-bc, 2023-08-07:
https://github.com/fprime-community/fprime-system-reference/tree/dev/actuator

matt392code, 2025-01-09:
Proposed PWM manager for controlling the rotor:
[pwm-manager.txt](https://github.com/user-attachments/files/18358515/pwm-manager.txt)
Proposed implementation file:
[pwm-manager-impl.txt](https://github.com/user-attachments/files/18358534/pwm-manager-impl.txt)
This PWM manager implementation provides:
1. SPWM (Sinusoidal Pulse Width Modulation) control
2. Configurable carrier and reference signal amplitudes
3. A default modulation index of 0.8 
4. The ability to control the rotor angle through the `setAngle_handler`
5. Direct duty cycle control through `setPWMDuty_handler` if needed
The component uses:
- Triangle carrier wave generation
- Sinusoidal reference signal
- Comparison-based pulse generation
- Configurable PWM frequency (default 10kHz)

To use this in the F Prime framework, you would need to:
1. Add this component to your topology
2. Connect the appropriate ports for angle control
3. Configure the hardware interface for PWM output