nasa/fprime #2120: F' running on RISC-V architecture

ethancheez, 2023-06-28:
Deploy F' on RISC-V. This can use the Zephyr OS layer

thomas-bc, 2023-09-19:
@LeStarch tagging you because I believe you have updates on this ?

ethancheez, 2023-09-20:
> @LeStarch tagging you because I believe you have updates on this ?

There are some RISC-V boards supported by ZephyrRTOS. And since I did get a zephyr deployment working, technically it should work with RISC-V. I just don't have a RISC-V board to test with.

LeStarch, 2023-09-21:
We got this running on a MangoPI (RISC-V 64bit) running Linux as well as a simulated RISC-V 64-bit Linux machine.  @ethancheez  is spot-on, the OS does the heavy lifting.

The process was as simple as compiling the standard F´  Ref deployment for Linux  using the RISC-V compiler.  Zephyr would be the same.

Closing as this is accomplished.

closed 2023-09-21 by LeStarch