nasa/fprime #2124: Math Component Tutorial Concerns and Suggestions

John-P-Cargille, 2023-07-05:
Going through the Math Component tutorial, I've noticed some things I think could be improved upon, or which I at least have some questions about.

1. In 4.5.2, a line reads "Add a helper function: Go into the directory Ref/MathSender/test/ut. In the file Tester.hpp, add the following lines to the section entitled "Helper methods"..." However, that section wasn't auto-generated as a header for me and both of the peers I checked with, as "Tests" was in Tester.cpp. So I assume it either should be changed to also auto-generate that header or the direction should ask you to put it the header yourself.

2. In 6.1, when directed to define `instance mathSender: Ref.MathSender base id 0x0E00 \` the base id overlapped with `instance typeDemo: Ref.TypeDemo base id 0x0E00` for me. Perhaps this was because my block driver was automatically given the base id 0x0100 as opposed to 0x0000: `instance blockDrv: Drv.BlockDriver base id 0x0100 \`. If this happens for other people, it would be worth changing the base id for mathSender to `0x0F00`.

3. In 6.2, when directed to "Packetize the telemetry channels", both packets close with the line `<packet/>`. This not only does not resemble the syntax of the packets written around it, but also actively produces errors when you later try to build the project. I assume it should be changed, as my peers and I all had to change it ourselves (to `</packet>`) before moving on.

4. Also in 6.2, at "Check for connected ports", it may help to put a short explanation as to what the command `fprime-util fpp-check -u unconnected.txt` does. I somehow managed to create a file simply called "unconnected" and then wasn't sure how it got there and if it was related to the previous command. Definitely a small issue, but adding a little blurb saying "this will write the unconnected ports into a file, which we'll then print to the terminal screen" would help anyone who does similar, especially because it's an F-Prime-specific command.

5. Also in 6.2, at "Connect mathReceiver to rate group 1", it's a bit unclear as it stands why you don't need to add a specific array index to the line `rateGroup1Comp.RateGroupMemberOut -> mathReceiver.schedIn` after "RateGroupMemberOut", and is something I'm still not certain on. This might be worth clarifying.

6. In 6.4, the current commands you're directed to enter to get the F Prime Visualizer tool to work are `echo DATA_FOLDER=Ref/ > .fpv-env \ nodemon [path to fpv root]/server/index.js ./.fpv-env`. However, providing the tool with "DATA_FOLDER=Ref/" doesn't actually seem to provide it with a valid path. The path I and my peers needed to use was one that started explicitly from the root directory and went to the "visual/Ref/" directory (i.e. `/home/[user]/.../[ProjectName]/fprime/Ref/Top/visual/Ref/`). This was especially confusing given that we didn't know exactly how the FPV tool works, and so at first assumed the tool would autofill the rest of the path somehow.

7. In 7.0, the tutorial currently assumes that the gds interface will automatically open when you run `fprime-gds` and fails to inform the user that they can manually navigate to it by going to "http://localhost:5000". I don't think I've ever personally had the site open automatically, and if I hadn't done it a number of times in the past this may have caused me some confusion.

Thanks for taking the time to interact with my concerns; hopefully this feedback will prove helpful as you continually update and improve your F Prime tutorials!

John-P-Cargille, 2023-07-11:
Awesome. Glancing over the newer version, I can definitely find things that seem intuitively better, as well as some helpful best practices and ideas I wouldn't have gotten to on my own. Hope the updating goes smoothly!

thomas-bc, 2023-07-25:
I'm going to go ahead and close this - feel free to re-open this or a new issue if you have any other suggestion!

closed 2023-07-25 by thomas-bc

John-P-Cargille, 2023-07-27:
> I'm going to go ahead and close this - feel free to re-open this or a new issue if you have any other suggestion!

That's totally fine; would you mind giving me some advice on when I should and should not close out issues, or letting me know if there somewhere where those best practices are recorded? For instance: I had another issue, which was a bug which was ultimately different than what I initially thought and largely solvable, and thus closed it out in a bit of panic. However, the issue still revealed a point of misunderstanding, an incompatibility between two adjacent versions of fprime-gds, and let them know that I had to revert to a previous version of the "flask" python tool to launch the gds in question--all information which may have been useful to have a few JPL eyes scan over before it got taken away from the top of the page. Here's [the closed issue I'm talking about](https://github.com/nasa/fprime/issues/2145), if you wanted to reference it directly. Thanks!

thomas-bc, 2023-07-27:
No worries at all. On the user side, I would close an issue when that specific issue is resolved with a PR, or when the question is understood and answered.
Better keep an issue open too long than close it before it's actually fixed, thus loosing track of it. I sometimes scroll through issues to close out things that have been resolved, so no worries about not closing out your own issue.

John-P-Cargille, 2023-07-28:
Okay, thanks for the clarification! In the future, I'll avoid closing out issues too quickly if they have loose ends, to make sure people see the pieces of them that might still be noteworthy.