nasa/fprime #2125: A Small Error and a Question in the FPP User's Guide

John-P-Cargille, 2023-07-05:
1. In 9.4, after the first two code blocks, the next sentence currently reads "When the implementation of ExternalSelfMessage invokes the port pInternal, a message goes on its queue. This corresponds to the behavior of pOut in ExternalSelfMessage". However, looking at the code blocks above, it seems to make much more sense for this to read "When the implementation of **InternalSelfMessage** invokes the port pInternal, a message goes on its queue. This corresponds to the behavior of pOut in ExternalSelfMessage".

2. Throughout 9.5, the `cmdRegOut` port is consistently placed before the `cmdIn` port, even though command input comes first chronologically and the order is swapped in most other material F Prime material I've encountered; this includes both earlier in the FPP user guide, as well as in other tutorials I've completed. So, while I can't say for certain if this is mistake, and it wouldn't cause errors if written in an actual F Prime project, I figured I'd point out the inconsistency in case it was unintentional.

thomas-bc, 2023-07-06:
@John-P-Cargille thanks for the feedback. Feel free to open a PR to correct these in https://github.com/fprime-community/fpp - we welcome contributions! 🙂

LeStarch, 2023-07-10:
Point 1, I think you are correct.  Point 2, the order should not matter....but consistency does.

John-P-Cargille, 2023-07-11:
Sounds good. I'll try to create a separate pull request for each issue (I'm still a bit new to this) so that you can accept the first change while still resolving the consistency issue differently if you so desire to.

bocchino, 2023-07-13:
> cmdRegOut port is consistently placed before the cmdIn port, even though command input comes first chronologically

Command registration happens before command input. I think that's the reason for the ordering here. I wouldn't worry too much about this ordering though.

John-P-Cargille, 2023-07-14:
Ah, apologies, that was my misunderstanding. Coincidentally, a contributor helped guide me to the User's Guide section on commands, events, telemetry, and parameters, and I now have a better understanding of the timeline when it comes to command calls. I would still offer that--while I don't know if this would be true for everybody--the ordering being the other way earlier in the fpp guide may have actually contributed to my confusion, so it might be worth making it consistently chronological so as to lessen the chance someone else makes the same mistake.

John-P-Cargille, 2023-07-14:
Oh, seems like you accepted both the commits from my pull request, and made it consistently cmdIn first across the fpp user guide. My last comment still stands as a small personal concern, but if y'all think the ordering of the ports is of little consequence then I'm sure it's fine one way or the other; people can always consult other material if they're confused, after all. In light of that, I'll go ahead and close out this issue. Thanks for looking at that for me!

closed 2023-07-14 by John-P-Cargille

bocchino, 2023-07-14:
OK, sounds good. I'll add some text to  § 9.3.1 of the FPP User's Guide explaining what the command ports do. I think that will dispel any remaining confusion.