nasa/fprime #2135: Add offset option for rate groups allowing different trigger patterns

csmith608, 2023-07-17:
| | |
|:---|:---|
|**_F´ Version_**| baremetal |
|**_Affected Component_**| RateGroupDriver  |
---
## Feature Description

Addition of an offset option when specifying rate group dividers. The offsets would be used to stagger RG firings, this is useful when a thread scheduler is not present

## Rationale

This was used on the LFPS firmware and cut our rate group times almost in half preventing major rate group slipping. The image below shows an example of how the current rate group would fire off vs how the proposed change could be used to fire them off.
This is an example of how the rate groups fired before and after adding the offsets. 
[image]

It was implemented [here](https://github.com/nasa/fprime/pull/818) for fprime 3.0.0, but should have used a struct for the rate group divider and offset. However, the logic changes are there and can be used pretty much as is. [Here](https://github.com/nasa/fprime/pull/818#pullrequestreview-706694500) are the relevant requested changes by @LeStarch 

@ethancheez

SMorettini, 2023-07-28:
Hello @csmith608 , I need this feature too, are you already implementing it? Otherwise, I'll work on it next week.

csmith608, 2023-07-30:
Hi @SMorettini, I'm excited that this feature is immediately useful. I don't have the bandwidth right now to do the implementation. I was hoping @ethancheez was going to have a chance, but I'm not sure, @LeStarch?

SMorettini, 2023-08-01:
@csmith608 I create the [pull request](https://github.com/nasa/fprime/pull/2166)  since I had to implement it on my private repository as soon as possible.

thomas-bc, 2023-10-26:
Merged! 🚀

closed 2023-10-26 by thomas-bc