nasa/fprime #2139: Unable to run HelloWorldDeployment with `fprime-gds`

capsulecorplab, 2023-07-18:
| | |
|:---|:---|
|**_F´ Version_**| 3.2.0 |
|**_Affected Component_**| HelloWorld |
---
## Problem Description

Encountered `[ERROR] Not in fprime deployment and no artifacts root provided, unable to find dictionary and/or app` while attempting to run `fprime-gds` after building HelloWorldDeployment from [fprime-tutorial-hello-world](https://github.com/fprime-community/fprime-tutorial-hello-world) as of https://github.com/fprime-community/fprime-tutorial-hello-world/commit/7958de6287ea7f88aa50a963a46937bccda3fe98

[image]

[image]

## How to Reproduce

### Setup dev environment
1. `git clone https://github.com/nasa/fprime.git`
2. `cd fprime`
3. `git checkout v3.2.0`
4. `pip install -r requirements.txt`

Note: Above steps are pre-installed/configured in [fprime-workspace-image](https://github.com/capsulecorplab/fprime-workspace-image)

### Build and attempt to run HelloWorldDeployment
1. `git clone --recurse-submodules https://github.com/fprime-community/fprime-tutorial-hello-world`
2. `cd fprime-tutorial-hello-world`
3. `git checkout 7958de6287ea7f88aa50a963a46937bccda3fe98`
4. `fprime-util generate`
5. `fprime-util build -j4`
6. `cd HelloWorldDeployment`
7. `fprime-util build -j4`
8. `fprime-gds`
9. See error

## Expected Behavior

Should be able to run deployment

ethancheez, 2023-07-20:
The new HelloWorld tutorial was completed using a new version of fprime-tools (currently unreleased).

The new tools creates a new deployment without `settings.ini` with the intention that you can now generate/build in the project root directory instead of in the `HelloWorldDeployment` directory.

You should be able to run `fprime-gds` in the project root directory (`fprime-tutorial-hello-world`) instead of in `HelloWorldDeployment`

capsulecorplab, 2023-07-23:
Tried re-generating and re-building in the project root (`fprime-tutorial-hello-world`), but still unable to run `fprime-gds`
[image]

thomas-bc, 2023-07-25:
Hi @capsulecorplab - thank you for opening this issue! We are aware of this, and it is fixed in the `devel` version of the fprime-gds. The tutorials were built with the newest-and-greatest tool suite in mind, this is why things are a little out of sync for now. In the meantime you can install the `devel` version of the tools, and the upcoming release (within a week or so) will include everything needed. Apologies for the inconvenience.

thomas-bc, 2023-08-03:
Closing this issue as it should be fixed by using the latest v3.3 release. Feel free to comment/reopen should you have issues getting it working.

closed 2023-08-03 by thomas-bc