nasa/fprime #2145: Experiencing Problems Launching the GDS in a Cloned Project on a Separate User-Name (and Machine)

John-P-Cargille, 2023-07-20:
| | |
|:---|:---|
|**_F´ Version_**|  fprime-tools v3.2.0; fprime-gds v3.2.1a1
|**_Affected Component_**| fprime-gds and cloned repos
---
## Problem Description

Some context for this problem, albeit from a previous point of understanding, can be found in [this](https://github.com/nasa/fprime/issues/2142) issue. The short version is that my C&DH lead created a new F' project, generated and built it, made a components folder which was empty except for a README.txt, and pushed all of these changes to a team repository. When the lead cloned the repository, they were able to create deployments and run `fprime-gds` with no issues. When the rest of the team attempted to do the same, we got at error which looked similar to the following:

[image]

We found this to be true regardless of our linux distribution, whether or not we were in the project's virtual environment (although interestingly I learned that, when using a different project's virtual environment, one could skirt around this issue and still access the correct project through the gds), our version of fprime-tools (between 3.2.0 and 3.2.1), our version fprime-gds (between 3.2.1a1 and 3.2.0), and what branch of the repository we were on. Ultimately, my team lead suggest we try having me make a new user with the same user name as him, which--after all of the F' dependencies were installed--worked perfectly well. 
We had our suspicions that it might have been some hard-coded path dependent on the machine, but the username makes just as much sense. I made sure to check the F' User's Guide section on the F' development process, but couldn't find anything there about this. Is there a way to edit whatever is dependent on this username manually? This feels like something that would've come up for more people in the standard course of development, so maybe our project has extraneous factor we can't currently perceive.

## How to Reproduce

1. Have someone make an F' project, generate it, and build it.
2. Have the same person commit all changes to a publicly-accessible repository.
3. Have a different person clone the project, ensuring they are on an account with a different username than the person who made the project.
4. Have that second person create, generate, and build in a new deployment for the project.
5. Have that second person run the command `fprime-gds` within that deployment.

## Expected Behavior

The person trying to run the ground data system will get an error like the screenshot from above, informing them that a binary location does not exist.

dav-oss, 2023-07-20:
You sure /build-artifacts/Linux/bin/ exists in your project files?

dav-oss, 2023-07-20:
Make sure the person committing the project to repo includes the /build-artifacts/Linux/bin/ contents too.

closed 2023-07-21 by John-P-Cargille