nasa/fprime #2151: Clearer Terminal Messages for "fprime-gds" and "nodemon [visualizer]" Commands

John-P-Cargille, 2023-07-22:
| | |
|:---|:---|
|**_F´ Version_**| 3.2.0
|**_Affected Component_**|  GDS and visualizer activation
---
## Feature Description

Add a link to http://localhost:5000 as part of the message emitted in the terminal when `fprime-gds` is successfully run; add a note that the user can quit out using `Ctrl-C` as part of the message emitted when `nodemon [fprime-visual-path]/server/index.js ./.fpv-env` does the same.

## Rationale

I've noticed as I've worked with F' projects and tools more that `fprime-gds` tells the user that they can shut down the system by entering `Ctrl-C`, but that it doesn't give a link to the browser page where the GDS is running. I know that for some people it starts automatically; however, this has never happened for me personally, and I feel it would be a simple and worthwhile inclusion to the terminal message which accompanies a successful execution of the command.
Inversely, running the command to boot up the fprime visualizer gives a link to the browser page where it can be used, but does not inform the user of how to close out of said visualizer (using `Ctrl-C`). This isn't a huge deal, as `Ctrl-C` is a reasonably standard quit command from what I can tell, but it would help some newer users and make the two tools more consistent. 
I'll attach two screenshots below from the terminal--depicting a terminal in which I launched both the GDS and Visualizer earlier today in a recently-created project--for reference.

[image]

[image]

thomas-bc, 2023-07-24:
Agreed on all points! Just FYI, since you bring up the `nodemon` version of fprime-visual, please take a look at https://github.com/fprime-community/fprime-visual -- there is a new release of fprime-visual based on Flask, and released on PyPi, which should make the whole install/set-up easier.

> I know that for some people it starts automatically; however, this has never happened for me personally

I'm curious, what system are you running on? And what browsers do you have installed? `fprime-gds` uses Python's `webbrowser.open()` to open a browser - maybe your system is having an issue with it for some reason. Some googling might help solve it.

John-P-Cargille, 2023-07-28:
The F Prime Visualizer is definitely much cleaner as a pip package, in terms of both instillation and usability. And it preempted my requested addition of the "Ctrl+C to quit" message, so that's no longer a concern!
As to why my localhost page doesn't open, I'm running Windows 11 as my os and chrome as my browser, using WSL-based Ubuntu as the virtual environment for my actual F' projects. Most of the stuff I can find on webbrowser.open() seems to assume I have direct access to the python script which is using said command--is this something that's exposed in F' projects? I'm not wholly sure how to find where I could alter that manually in my code. The other recommendation I saw that seemed somewhat promising is to make sure webbrowser.open() wasn't being blocked by a windows firewall, but I'm also not fully sure how I would find where I find what port the gds is trying to go through.

thomas-bc, 2023-09-06:
Should also be addressed by this ticket:
- `fprime-util visualize` should launch/display message

closed 2024-03-28 by thomas-bc