nasa/fprime #2156: Failed to build component

foxhoundsk, 2023-07-26:
| | |
|:---|:---|
|**_F´ Version_**| 3.2.1 |
|**_Affected Component_**| n/a |
---
## Problem Description

During the deployment step in the tutorial, I got the following error:
```
/home/fox/fprime/MyProject/MyDeployment/Top/instances.fpp: 85.17
  instance fox: Components.MyComponent base id 0x0F00 \
                ^
error: undefined symbol Components
gmake[3]: *** [MyDeployment/Top/CMakeFiles/MyDeployment_Top.dir/build.make:112: MyDeployment/Top/MyDeploymentTopologyAppAi.xml]
Error 1
gmake[2]: *** [CMakeFiles/Makefile2:8240: MyDeployment/Top/CMakeFiles/MyDeployment_Top.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:2002: CMakeFiles/MyDeployment.dir/rule] Error 2
gmake: *** [Makefile:195: MyDeployment] Error 2
[ERROR] CMake erred with return code 2
```

However, the directory, `Components`, was presented in the project's root directory.

A workaround for this is to append the following to `CMakeLists.txt` in the deployment directory (`MyDeployment/` in my case):
```
add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/../Components/MyComponent")
```

I guess command `fprime-util new --deployment` should populate the `CMakeLists.txt` properly, but for some reason it didn't. Any thoughts?

Thanks!

## How to Reproduce

Follow the [official tutorial](https://fprime-community.github.io/fprime-tutorial-hello-world).

## Expected Behavior

A Successful build.

thomas-bc, 2023-07-26:
Hi @foxhoundsk ! My guess is that you're using the latest release of `fprime-tools`? The tutorial is built around the `devel` version of fprime-tools to generate the correct templates with `fprime-util new ...`. Specifically, the CMakeLists.txt have been modified.
Apologies for the inconvenience, we are coming up with a new release soon (should be within a week) and we didn't want to mess with the structure of the tutorial for a short period of time. If you look at the reference in [fprime-tutorial-hello-world](https://fprime-community.github.io/fprime-tutorial-hello-world), you can see what the CMakeLists.txt files look like in the most modern version. The `project()` call should be removed from the deployment/CMakeLists.txt along with some other modifications.
But your workaround of including `add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/../Components/MyComponent")` will work too!

foxhoundsk, 2023-07-27:
> Hi @foxhoundsk ! My guess is that you're using the latest release of `fprime-tools`? The tutorial is built around the `devel` version of fprime-tools to generate the correct templates with `fprime-util new ...`. Specifically, the CMakeLists.txt have been modified.

Yep, I use the one installed with pip, which has ver. `3.2.1`.

> Apologies for the inconvenience, we are coming up with a new release soon (should be within a week) and we didn't want to mess with the structure of the tutorial for a short period of time. If you [...]

Got it, thank you for the info!

closed 2023-07-27 by foxhoundsk