nasa/fprime #2157: Update Seqgen to support complex commanding

thomas-bc, 2023-07-26:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Update GDS Seqgen to support complex types commanding

LeStarch, 2023-10-24:
Did some initial investigation on this.  

1. This is going to be a substantial change as most of the seqgen file parser needs to be replaced
2. The algorithm the UI uses has been ported to python here: https://github.com/LeStarch/fprime-gds/tree/update/complex-sequencing
3. Need to deal with the time tokens:
  1. Could parse separately
  2. Could add a regex clause to support R and A time tokens
4. Need to deal with `;`

LeStarch, 2024-03-06:
@Joshua-Anderson and @timcanham given the desire to improve the sequencer, is improving the old sequencing tools still valuable?  Or should I shelve this issue as it will be resolved with the revitalization effort?

matt392code, 2025-01-09:
[seqgen-analysis.md](https://github.com/user-attachments/files/18358664/seqgen-analysis.md)
**Recommendations**: 
   - Proceed with the implementation but at a measured pace
   - Focus on making the changes modular to potentially benefit the future revitalization effort
   - Ensure backward compatibility is maintained
   - Add comprehensive testing for complex type scenarios

closed 2025-10-01 by thomas-bc