nasa/fprime #2161: Add OpenMCT Telemetry Viewer to F-Prime GDS

mohitsingh999, 2023-07-26:
| | |
|:---|:---|
|**_F´ Version_**| 3.2.1|
|**_Affected Component_**|  |
---
## Feature Description
An OpenMCT Telemetry viewer in F-Prime GDS would involve creating a plugin that would:

1. Take in the TopologyAppDictionary.xml and generate OpenMCT JSON Definitions
2. Poll the F-Prime GDS API for telemetry and publish the data to an OpenMCT server

Ideally, the user would just run the command ```fprime-gds``` and an OpenMCT server would start as a separate tab, along with the standard F-Prime GDS server. The user would get access to all the F-Prime telemetry channels in the OpenMCT server, which would display the channels in real time.

## Rationale

Currently, the way to view telemetry in F-Prime GDS is via the charts tab and the channels tab. The current telemetry views allow for raw timestamp viewing and individual plots of telemetry. OpenMCT allows for additional customizability with telemetry viewing. Users can create layouts with multiple telemetry channels, and visualize telemetry in a much more refined manner(stacked/overlay plots, gauges, tables).

thomas-bc, 2023-09-06:
We may want to make this rely on the results of https://github.com/nasa/fprime/issues/2268 if possible

thomas-bc, 2024-03-11:
We now have a plugin architecture for the GDS: https://github.com/fprime-community/fprime-gds/pull/164 🎉 

I am marking this as `help-wanted`. If anyone from the community would like to take a pass at this, we have a pilot implementation here: https://github.com/fprime-community/fprime-gds/pull/136

The OpenMCT integration could be spun up into a plugin that we could publish in fprime-community.

matt392code, 2025-01-09:
Proposed analysis of how to implement OpenMCT as a telemetry viewer for F-Prime GDS:
[openmct-fprime-integration.tsx.txt](https://github.com/user-attachments/files/18358761/openmct-fprime-integration.tsx.txt)
1. Required Components:
- Dictionary Plugin:
  - Parse TopologyAppDictionary.xml from F-Prime
  - Generate OpenMCT dictionary format objects
  - Create telemetry point definitions matching F-Prime channels

- Telemetry Plugins:
  - Historical telemetry adapter to fetch past data from GDS API
  - Realtime telemetry adapter using WebSocket connection
  - Time system configuration matching F-Prime timestamps

2. Integration Steps:

```javascript
// Example implementation outline
function FPrimeGDSPlugin() {
    return function install(openmct) {
        // Register telemetry dictionary from XML
        registerTelemetryDictionary(openmct);
        
        // Setup historical telemetry provider
        const historicalProvider = {
            supportsRequest: (domainObject) => {
                return domainObject.type === 'fprime.telemetry';
            },
            request: async (domainObject, options) => {
                // Query F-Prime GDS API for historical data
                return await fetchGDSHistory(domainObject.identifier.key, 
                                          options.start, 
                                          options.end);
            }
        };
        
        // Setup realtime telemetry provider
        const realtimeProvider = {
            supportsSubscribe: (domainObject) => {
                return domainObject.type === 'fprime.telemetry';
            },
            subscribe: (domainObject, callback) => {
                // Connect to F-Prime GDS WebSocket
                return subscribeGDSChannel(domainObject.identifier.key, callback);
            }
        };
        
        openmct.telemetry.addProvider(historicalProvider);
        openmct.telemetry.addProvider(realtimeProvider);
    }
}
```
3. Key Benefits:
- Enhanced Visualization Options:
  - Multiple visualization types (plots, gauges, tables)
  - Customizable layouts
  - Overlay/stacked plots
  - Time-synchronized views

- Improved User Experience:
  - Modern web interface
  - Mobile compatibility 
  - Drag-and-drop layout customization
  - Integrated health monitoring

4. Implementation Strategy:
1. Create GDS plugin using new plugin architecture (nasa/fprime-gds#164)
2. Develop XML to OpenMCT dictionary converter
3. Implement telemetry adapters for GDS API
4. Package as fprime-community plugin
5. Add auto-launch with fprime-gds command

5. Technical Considerations:
- Ensure proper timestamp handling between systems
- Handle different telemetry data types
- Implement efficient data streaming
- Support F-Prime's specific channel configurations
- Maintain synchronized views across components

This integration would significantly enhance F-Prime's telemetry visualization capabilities while leveraging OpenMCT's proven mission operations heritage at NASA/JPL.

matt392code, 2025-01-09:
[image]