nasa/fprime #2179: fprime-util new --project doesn't create .gitignore

timcanham, 2023-08-03:
| | |
|:---|:---|
|**_F´ Version_**| 3.x|
|**_Affected Component_**| n/a  |
---
## Problem Description

The tutorial to create a new deployment does not place a `.gitignore` file in the `MyProject` folder. This risks checking in the `venv` folder that gets created by default in `MyProject`.

## How to Reproduce

1. Follow `Hello World` tutorial through the new project step
2. Notice that `.gitignore` is missing

## Expected Behavior

A `.gitignore` should be created with `venv` in it, or better the location of the user-selected virtual environment if it is in the newly-created repo.

thomas-bc, 2023-08-03:
Argh, it seems this is a bug with cookiecutter (https://github.com/cookiecutter/cookiecutter/issues/1797), because there is a .gitignore in our project template ([.gitignore](https://github.com/fprime-community/fprime-tools/blob/devel/src/fprime/cookiecutter_templates/cookiecutter-fprime-project/%7B%7Bcookiecutter.project_name%7D%7D/.gitignore)). We'll have to find a workaround with the post-generation scripts.

thomas-bc, 2023-08-03:
It was actually a little more subtle than that, see the above PR for explanation

closed 2023-08-03 by LeStarch