nasa/fprime #2190: Add Spacecraft Id to Command Sequences

Joshua-Anderson, 2023-08-14:
| | |
|:---|:---|
|**_F´ Version_**| v3.3 |
|**_Affected Component_**| CmdSequencer, seqgen, dictionary |
---

## Feature Description

Add the concept of an integer (U32/U64) spacecraft id to F', which would then be included in the generated dictionary and as a C++ constant.

This spacecraft id would then be added to the header of compiled command sequences.
The command sequencer would reject command sequences that with a spacecraft id mismatching the deployment.

## Rationale

Executing command sequences compiled with incorrect dictionary can cause unintended spacecraft behavior. Since the F' sequence format is standardized, there's nothing preventing executing a sequence intended for one deployment for fprime on another deployment.

A U64/U32 spacecraft id a large enough identifier space to hopefully protect against unintentional execution of sequences intended for different deployments.

The spacecraft id can also be updated as desired when making backwards incompatible command changes, preventing execution of sequences compiled for an old FSW version.

Because this identifier will go in binary command sequences a string identifier unfortunately seems too large, so I'm suggesting making spacecraft id an integer.

matt392code, 2025-01-09:
Proposed header file for the spacecraft ID integration in F Prime:
[spacecraft-id.txt](https://github.com/user-attachments/files/18359019/spacecraft-id.txt)
Key features include:

1. SpacecraftId class:
   - Uses U64 for maximum flexibility
   - Includes reserved IDs for testing and emergency override
   - Static methods for ID validation and conversion

2. SpacecraftIdSerializer class:
   - Handles serialization/deserialization for command sequences
   - Integrates with F Prime's existing serialization framework
   - Provides buffer size calculation for memory management

3. Safety features:
   - Reserved ID values for special cases
   - Validation methods to prevent misuse
   - String conversion for logging and debugging

LeStarch, 2025-02-04:
@zimri-leisher as the driver behind our advanced sequencer, what do you think of this request?

zimri-leisher, 2025-02-05:
So, you're trying to mitigate the risk of user error: an operator erroneously compiles a sequence intended for deployment A against a dictionary from deployment B, and it calls unintended commands when uploaded to the spacecraft.

I think there may already be sufficient mitigation of this risk. Sequences will fail to compile if instance/command/module names are different, or if their types are different. This means if the sequence compiled successfully for B, B has the same command names in the same component instance names with the same, or very similar, argument types. 

In this scenario, the erroneous sequence could only erroneously call commands with the same name, owning component instance and arg types. If the deployments are already being confused with each other, it's much more likely than not that these are indeed the exact same commands.

However, this doesn't address if the opcodes of the commands change between deployments in an unlucky way (a "START_BURN" opcode swapped with an "ABORT_BURN" command, e.g.). This does seem extremely unlikely, as there would have to be two different components with the same instance name, with reversed opcodes... What's much more likely is that, instead of this happening between two different deployments, it happens due to a versioning mismatch between ground and spacecraft. However, then this "deployment ID" no longer prevents the problem. It seems like you'd want to verify what commit your FSW is running, and what commit your dictionary was generated on. I think maybe this is best solved on the ops side, not via code.

Joshua-Anderson, 2025-02-06:
> I think maybe this is best solved on the ops side, not via code.

Yes, the risk of compiling with the wrong dictionary should primarily be solved on the ground side.

Despite best intentions however, I have seen attempted execution of an incompatible sequence on multiple (!) missions. Whether it's forgetting to clear incompatible sequences after a FSW update, executing a sequence on the wrong node of a multi-deployment system, they're mistakes that ground tools can mitigate but are nearly impossible to entirely protect against in all the possible edge cases. Even during the development phase when folks are testing on testbeds it's nice to protect testbed hardware by ensuring that the sequences used for testing are compatible with whatever the latest testbed software build is.

I agree that the odds of an invalid sequence triggering unwanted behavior is low, and typically you'll fault pretty quickly on a invalid opcode or arguments, but you could definitely crash the flight software or fault into a safe mode, which could have undesired consequences. This risk is actually greater with F' missions, since there's a lot more common components that are shared between missions, so there's a greater probability of common commands and opscodes.

From my view, it's a very low cost check to add an spacecraft id/dictionary id to your sequence file and can help provide an clear error if (or when) this occurs, as well as providing an easy way to inspect and determine if a compiled sequence file was compiled with the expected dictionary/for the intended target, which is otherwise difficult to do since all F' sequences will otherwise look the same.

timcanham, 2025-02-06:
FWIW, I agree with embedding a dictionary ID in the sequence file. It should be relatively straightforward, but it should be automated and not dependent on a user remembering to update it. I had a concept rattling around in my head where the build tools could do an automatic hash of the dictionary file itself, and that could act as an ID. The tricky part would be to get this ID back into the build as a constant that could be used by the sequencer as a check. We could use the git hash of the software release, but that could cause a lot of thrash when there were incremental updates to the software that didn't change the dictionary.

zimri-leisher, 2025-02-07:
Okay, I trust both of your experience a lot so I would now lean towards adding this. I just push back on new requirements as a matter of course, frequently they give way. I think this will only linearly increase complexity of the sequencer though, so it's not so bad.

I think the hash of the dictionary file is the best solution here. We may want some option to disable this check, because it could be problematic if you can't run a sequence that you know will work, and you're unable to upload a new one for some reason.

Joshua-Anderson, 2025-02-07:
> We may want some option to disable this check, because it could be problematic if you can't run a sequence that you know will work

I'd consider just provide an option in the sequence compiler to manually set this id when compiling the sequence, avoiding the need to bypass in the FSW