nasa/fprime #2196: fprime-util build failed on `error: ‘Fw::InputCmdPort’`

iftahnaf, 2023-08-15:
| | |
|:---|:---|
|**_F´ Version_**| 3.3.0|
|**_Affected Component_**|  FW |
---
## Problem Description
when trying to compile a project with `fprime-util build` it fails with error:

```
 fprime-util build
[  0%] Built target proto
[  4%] Built target codegen
[  4%] Built target config
[  4%] Built target Fw_Cfg
[ 10%] Built target Fw_Types
[ 10%] Built target Fw_Logger
[ 12%] Built target Fw_Obj
[ 12%] Built target Fw_Port
[ 14%] Built target Fw_Time
[ 14%] Built target Fw_Com
[ 16%] Built target Fw_Tlm
[ 20%] Built target Fw_Log
[ 22%] Built target Fw_Cmd
[ 24%] Built target Fw_Prm
[ 24%] Built target Fw_Buffer
[ 26%] Built target Utils_Hash
[ 26%] Built target Fw_Comp
[ 34%] Built target Os
[ 34%] Built target Fw_CompQueued
[ 36%] Built target Fw_Ports_SuccessCondition
[ 36%] Built target CFDP_Checksum
[ 38%] Built target Fw_FilePacket
[ 40%] Built target Svc_Cycle
[ 40%] Built target Svc_Fatal
[ 40%] Built target Svc_Ping
[ 42%] Built target Svc_Sched
[ 44%] Built target Svc_Seq
[ 46%] Built target Svc_WatchDog
[ 48%] Built target Svc_ActiveLogger
[ 51%] Built target Svc_ActiveRateGroup
[ 53%] Built target Svc_AssertFatalAdapter
[ 55%] Built target Svc_BufferManager
[ 57%] Built target Svc_CmdDispatcher
[ 61%] Built target Svc_CmdSequencer
[ 63%] Built target Utils_Types
[ 63%] Built target Svc_FramingProtocol
[ 65%] Built target Drv_ByteStreamDriverModel
[ 65%] Built target Svc_Deframer
[ 67%] Built target Svc_FatalHandler
[ 69%] Built target Svc_FileDownlinkPorts
[ 71%] Built target Svc_FileDownlink
[ 73%] Built target Svc_FileManager
[ 75%] Built target Svc_FileUplink
[ 75%] Built target Svc_Framer
[ 77%] Built target Svc_Health
[ 79%] Built target Svc_PrmDb
[ 81%] Built target Svc_RateGroupDriver
[ 81%] Built target Svc_StaticMemory
[ 83%] Built target Svc_Time
[ 85%] Built target Svc_TlmChan
[ 85%] Built target version
[ 87%] Built target Svc_SystemResources
[ 87%] Built target Svc_PassiveConsoleTextLogger
[ 87%] Built target Svc_LinuxTime
[ 87%] Built target Drv_DataTypes
[ 87%] Built target Drv_BlockDriver
[ 87%] Built target Drv_Ip
[ 89%] Built target Drv_TcpClient
[ 91%] Built target Drv_Udp
[ 91%] Built target spring_spring_types
[ 91%] Built target spring_spring_ports
Scanning dependencies of target spring_controller
[ 93%] Building CXX object spring/controller/CMakeFiles/spring_controller.dir/controller_comp.cpp.o
[ 93%] Building CXX object spring/controller/CMakeFiles/spring_controller.dir/controller_compComponentAc.cpp.o
/fprime_ws/spring/build-fprime-automatic-native/spring/controller/controller_compComponentAc.cpp:26:28: error: ‘Fw::InputCmdPort’ has not been declared
   26 |       BYTE cmdPortSize[Fw::InputCmdPort::SERIALIZED_SIZE];
```

## How to Reproduce

1. fprime-util generate
2. fprime-util build

anything specific that I need to take care with this kind of error?

Thanks in advance.

LeStarch, 2023-08-16:
Make sure all three command ports are defined in the `spring_controller` FPP model:

```
    @ Command registration port
    command reg port cmdRegOut

    @ Command received port
    command recv port cmdIn

    @ Command response port
    command resp port cmdResponseOut
```

All three are required when using commands.

iftahnaf, 2023-08-16:
thanks, for some reason the `command reg port cmdRegOut` was commented out.