nasa/fprime #2203: Run CI on UH Cubesat

thomas-bc, 2023-08-16:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

https://github.com/fprime-community/fprime-artemis-cubesat/tree/dev

To run on the Artemis Cubesat in CI

LeStarch, 2024-03-27:
@rlcheng this might be an extension of your work.

matt392code, 2025-01-09:
Proposed CI implementation:
[github-ci.txt](https://github.com/user-attachments/files/18359103/github-ci.txt)
This CI configuration:
1. Trigger Conditions:
- Runs on pushes to main and dev branches
- Runs on pull requests to main and dev branches

2. Environment:
- Uses Ubuntu latest with the official F' Docker base image
- Configures root user access for full build capabilities

3. Build Steps:
- Checks out code with submodules
- Installs necessary dependencies
- Generates and builds both Teensy and RPi deployments
- Combines the dictionaries
- Runs unit tests using F' testing framework

4. Artifact Storage:
- Stores built binaries for both Teensy and RPi
- Includes the combined dictionary file

To implement this:
1. Create a `.github/workflows` directory in your repository
2. Save the YAML configuration as `ci.yml` in that directory
3. Commit and push the changes

Additional considerations:
1. You may need to modify paths and build commands based on your specific UH CubeSat configuration
2. Consider adding static analysis tools like `clang-format` for code style checking
3. You might want to add deployment-specific tests
4. Consider adding environment-specific secrets for any sensitive configuration