nasa/fprime #2209: ComQueue Enhancements

gcgandhi, 2023-08-17:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  ComQueue |
---
## Feature Description

1. Ability to open/close/clear downlink queues 
2. Ability to change downlink rate.

## Rationale

These features have proven to be invaluable on large scale flagship missions and I anticipate being import for future larger F-prime projects.  Regularly used both in testing and OPs.

LeStarch, 2023-08-18:
@gcgandhi would you elaborate a bit more on these items? e.g. what is the function of new queues?

The current architecture ties the flow rate to the radio allowing the radio to control when new packets are sent.  What does downlink rate mean given this context?

gcgandhi, 2023-08-21:
1.  In flagship missions we can control whether DPs, EHA, EVRs are actually drained from their downlink queue.  Typically the DP queue is disabled by default and must be enabled by ground command.  In both OPs and testing it is useful to be able to control this, in order to not overwhelm downlink particularly in low rate scenarios or small comm passes.  Being able to clear I have only used in testing, but I anticipate they use it in OPs (but don't know the details).
2. Changing the downlink rate mainly involves having a ground command to be able to update the particular hardware appropriately.  Flagship missions have used a TIF FPGA that maintains a register for this, but it would obviously vary from mission to mission.  FSW also does various actions to gracefully change rates if trying to not lose data, etc.  So a lot of this is mission/hardware specific.

matt392code, 2025-01-09:
Proposed enhancements to the ComQueue component:
[comqueue-enhancements.txt](https://github.com/user-attachments/files/18368154/comqueue-enhancements.txt)

Explanation of the key features:

1. Queue Control Features:
- Added ability to open/close queues via `CMD_QUEUE_SET_STATE_handler`
- Added ability to clear queues via `CMD_QUEUE_CLEAR_handler`
- Implemented queue state tracking for different types (DP, EHA, EVR)
- EVR queue starts enabled by default, others start disabled
- Added activity logging for queue state changes

2. Downlink Rate Management:
- Added rate control via `CMD_SET_DOWNLINK_RATE_handler`
- Implemented rate validation and bounds checking
- Added hardware rate update mechanism (needs hardware-specific implementation)
- Added activity logging for rate changes
- Configurable min/max rate limits

The implementation includes:
- New enums for queue states and types
- Rate configuration structure
- Command handlers for all new operations
- Helper methods for validation and initialization
- Activity logging for operational visibility

To complete the implementation, you would need to:

1. Add the command/event definitions to the component XML
2. Implement the hardware-specific rate control in `updateHardwareRate()`
3. Add the queue clearing logic specific to your queue implementation
4. Add any necessary unit tests

matt392code, 2025-01-09:
Proposed new commands and events in F Prime's component XML format:
[comqueue-xml.txt](https://github.com/user-attachments/files/18368170/comqueue-xml.txt)

This XML component definition includes all the necessary elements to support our code implementation:

1. Commands:
   - QUEUE_SET_STATE: Controls queue open/closed state
   - QUEUE_CLEAR: Clears a specified queue
   - SET_DOWNLINK_RATE: Sets the downlink rate

2. Events:
   - QUEUE_STATE_CHANGE: Logs queue state transitions
   - QUEUE_CLEARED: Logs queue clear operations
   - DOWNLINK_RATE_CHANGE: Logs rate changes
   - INVALID_PARAMETER: Logs command validation failures

3. Telemetry Channels:
   - QueueDpState: Monitors Data Products queue state
   - QueueEhaState: Monitors Engineering Health & Status queue state
   - QueueEvrState: Monitors Event Reports queue state
   - CurrentDownlinkRate: Monitors current downlink rate

The XML definition aligns with F Prime's component framework and provides all the necessary interfaces for ground control and monitoring. The component will be able to:
- Receive and process commands from ground
- Generate events for state changes and operations
- Provide telemetry for monitoring queue states and downlink rate