nasa/fprime #2210: TlmPacketizer Enhancements

gcgandhi, 2023-08-17:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  TlmPacketizer  |
---
## Feature Description

1. TSRs- These are files that specify rules to determine which EHA to packetize for downlink and at what rate. Much more sophisticated than what currently exists in TlmPacketizer (i.e. simple priority or on demand).  Tim indicated this is already part of the plan for Prime.
2. Configurable EHA packet production rate 
3. EHA groups 
4. EHA OPs Cats (operational categories)

## Rationale

These features have been very valuable on large flagship missions in both testing and OPs.

matt392code, 2025-01-09:
Proposed enhanced design:
[tlm-packetizer-design.md](https://github.com/user-attachments/files/18368205/tlm-packetizer-design.md)

This design enhances the TlmPacketizer to support all requested features while maintaining F Prime's core principles. Key improvements include:

1. Sophisticated rule engine for dynamic telemetry handling
2. Flexible rate control at multiple levels (rule, group, category)
3. Hierarchical grouping for better organization
4. Operational categories for mission phase-specific behavior

The design maintains backward compatibility while adding these capabilities. Would you like me to elaborate on any specific aspect of the design?

Some key considerations I've included:
- Memory efficiency through pooling and static allocation where possible
- Performance optimization via caching and efficient data structures
- Clear migration path for existing systems
- Extensive testing strategy

matt392code, 2025-01-09:
Converted markdown to PDF:
[Enhanced TlmPacketizer Design Document.pdf](https://github.com/user-attachments/files/18368219/Enhanced.TlmPacketizer.Design.Document.pdf)

closed 2025-12-19 by thomas-bc