nasa/fprime #2211: Load and Go Sequences

gcgandhi, 2023-08-17:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  CmdSequencer |
---
## Feature Description

Implement Load and Go Sequences.  These are sequences that can run as soon as they are loaded onto the filesystem.

## Rationale

Used on large flagship missions and was also requested by system engineers on CADRE.

Joshua-Anderson, 2023-08-18:
My understanding is that load and go sequences don't require but pair well the concept of file uplink bundles, so you can deliver a bundle of files/sequences to the spacecraft, then use load and go to execute the main sequence of the bundle.

Otherwise it can be more challenging to ensure arrival of load-and-go sequence pre-reqs prior to start of the load and go.

rrieber, 2023-08-23:
I support this. I do want to mention that on M2020, the Load-and-go flag is buried in the "File Processing Field" of the "File Load Header". e.g. it's a flag in the file transfer layer, not in the sequence layer. I guess that might be appropriate.

But this feature is a huge timesaver. Traditionally, and operator would uplink a sequence, verify it's on board, then send an activate. Load-and-Go saves a round-trip-light time for activation.

And on CADRE, it's even more important. Our data can be be received on the rovers at 3Mbps. This means that if we chased a sequence load with an activate, there's a race condition between processing the file transfer and saving to disk and the reception of the activate immediate command. We have yet to test to see if this is a problem, but my hopes are not high.

TLDR, I second @gcgandhi's request!

Joshua-Anderson, 2023-08-23:
Hmm, that's a good point. Multiple file transfers back to back is inherantly a little less likely to be problematic than a file transfer followed by a execute command.

It would be pretty easy to add a load-and-go flag to the fileuplink/filedownlink file transfer header, and if that flag was set, FileDownlink could optionally (if connected) call a seq_execute_path port on receipt of an end file transfer packet.

gcgandhi, 2023-08-23:
@Joshua-Anderson I had the same idea as well when this was brought up by Rich:

So one implementation would be to update the File packet header to have this flag, may also need a flag to indicate its a sequence file.  Then have the FileUplink component call CmdSequencer to run the sequence if needed after getting the end file packet.  I estimate this would only take a couple days to implement/test.

matt392code, 2025-01-09:
FileUplink Mermaid diagram to visualize the FileUplink component's state machine and packet handling flow:
[fileuplink-state-diagram.mermaid.txt](https://github.com/user-attachments/files/18368268/fileuplink-state-diagram.mermaid.txt)

State diagram that shows:

1. The main states (START, DATA, END, CANCEL) and their transitions
2. Sub-states within each main state showing the detailed processing steps
3. Error conditions and warning paths
4. The complete lifecycle of file packet processing

Key aspects captured in the diagram:
- Initial START state handling and file opening
- DATA state processing including sequence validation and bounds checking
- END state with checksum validation
- CANCEL handling from any state
- All warning conditions mentioned in the documentation

The diagram reflects the documented assumptions about packet ordering and the component's error handling behavior. Each major state aligns with the packet types described in sections 3.5.1 through 3.5.4 of the documentation.

matt392code, 2025-01-09:
Proposed C++ class implementation that follows the NASA F Prime framework conventions and implements the FileUplink component:
[fileuplink-component.txt](https://github.com/user-attachments/files/18368283/fileuplink-component.txt)
Key features include:

1. State Management:
   - Tracks receive mode (START/DATA)
   - Maintains sequence indices
   - Handles file descriptors

2. Packet Processing:
   - Separate handlers for START, DATA, END, and CANCEL packets
   - Sequence validation
   - Checksum verification
   - Error handling

3. File Operations:
   - File opening/closing
   - Data writing with offset support
   - Checksum computation (placeholder for CCSDS implementation)

4. Telemetry & Events:
   - Warning emission
   - Statistics tracking
   - Telemetry output (commented placeholders)

5. Error Handling:
   - Invalid receive modes
   - Out-of-order packets
   - File I/O errors
   - Checksum mismatches

The code includes placeholders for F Prime-specific features like telemetry and event output ports. These would need to be properly integrated with the F Prime framework in a full implementation.