nasa/fprime #2223: Gap-identification vulnerability for file transfers

rrieber, 2023-08-23:
## Feature Description

Modify the "Sequence ID" field in the "File Packet Header" to be "Transaction ID" rather than a monotonically increasing counter. This would bring the protocol closer to the CFDP inspiration upon which it was written. The "Transaction ID" would enable the receiver to group packets of the same file together and permit the receiver to hold on to multiple file transactions that might be waiting for completion.

## Rationale

Currently, file packets are stamped with a "Sequence ID", which starts at 0 for the Start packet, increments to 1 for the first file data packet, then continues to count. This permits the receiver to order packets properly and identify gaps. However it is not robust.

### Gap Identification Failure
Let's explore an example where two files are telemetered that have the same size (which is a common activity) back to back. Each file has 4 packets (1 start, 2 data, 1 end).  Assume packets 3 and 4 of file-1 and packets 1 and 2 of file-2 are dropped. This would create the following packet assembly:
 - File-1, packet-1 (START)
 - File-1, packet-2 (DATA)
 - File-2, packet-3 (DATA)
 - File-2, packet-4 (END)
This would be happily reassembled, then fail checksum at file validation. 

### Dropped START Packet
Continuing the above example, the receiver would have zero clue that File-2 was sent in that it never received it's destination file name.  This would be a silent failure and the receiver would not be able to inform the sender of any evidence of this file.

### Redundant with Byte Offset/Data Length 
The ordering of file data packets is achieved with the byte offset and data length field and a sequence ID counter is not required.

## Conclusion
I have enumerated two issues why a unique transaction ID would be superior than a monotonic counter. I'm sure there are additional benefits to such a feature that I have missed. I do not think this would be an outlandish change in the protocol or the code. Hopefully this helps.

P.S. Maybe just deploy Type-2 CFDP and solve all the problems :-D

timcanham, 2023-08-23:
@rrieber I think the answer is both. We need a transaction ID along with a sequence ID so we can detect dropouts in the file transmission. The design of the FileDownlink/FileUplink has always been serial anyway - it doesn't simultaneously transmit more than one file.

As for CFDP, I think it would be a great idea to do it all around - we just need the resources to do it.

rrieber, 2023-08-23:
@timcanham You are correct in that files do not TRANSMIT more than one file at a time, but due to reception kerfuffles or auto-retransmits from different protocol layers (think PROX-1 or TCP/IP), files could be RECEIVED out of order.

Can you explain how sequence ID provides additional information beyond that of the byte offset/data length fields? It is my opinion that only the byte offset/data length is required to reassemble the file and identify gaps. (This is how CFDP does it.)

timcanham, 2023-08-23:
@rrieber Belt and suspenders, which I'm a fan of. It makes the receiver a little simpler to just detect a break in the sequence. We could also add offset and size checking to do full belt and suspenders.

matt392code, 2025-01-10:
Proposed mitigation strategies for the identified file transfer vulnerabilities in F Prime:

1. Transaction ID Implementation (Primary Fix):
```python
class FilePacketHeader:
    def __init__(self):
        self.transaction_id = uuid.uuid4()  # Unique ID per file transfer
        self.packet_type = None    # START/DATA/END/CANCEL
        self.byte_offset = 0       # Position in file
        self.data_length = 0       # Length of payload
```

2. Enhanced Packet Validation:
```python
class FileReceiver:
    def validate_packet(self, packet):
        # Verify packet belongs to expected transaction
        if packet.transaction_id not in self.active_transactions:
            raise InvalidTransactionError()
            
        # Verify byte offset continuity
        expected_offset = self.transactions[packet.transaction_id].next_offset
        if packet.byte_offset != expected_offset:
            self.request_retransmission(packet.transaction_id, expected_offset)
            
        # Validate packet size against file size (for START packets)
        if packet.type == PacketType.START:
            self.transactions[packet.transaction_id].total_size = packet.file_size
```

3. Transaction State Management:
```python
class TransactionState:
    def __init__(self):
        self.received_start = False
        self.received_chunks = set()  # Track received byte ranges
        self.file_size = None
        self.source_path = None
        self.dest_path = None
        
    def identify_gaps(self):
        # Return list of missing byte ranges
        missing = []
        current_pos = 0
        for chunk in sorted(self.received_chunks):
            if chunk.start > current_pos:
                missing.append((current_pos, chunk.start))
            current_pos = chunk.end
        return missing
```

4. Recovery Mechanisms:
```python
class FileTransferManager:
    def handle_missing_start(self, transaction_id):
        # Request START packet retransmission
        self.send_packet_request(
            transaction_id=transaction_id,
            packet_type=PacketType.START
        )
        
    def request_gap_fill(self, transaction_id, gaps):
        for start, end in gaps:
            self.send_data_request(
                transaction_id=transaction_id,
                byte_range=(start, end)
            )
```

5. Additional Safety Measures:

```python
class FilePacketValidator:
    def __init__(self):
        self.active_transactions = {}
        
    def validate_transaction_sequence(self, packet):
        transaction = self.active_transactions[packet.transaction_id]
        
        # Enforce packet ordering rules
        if packet.type == PacketType.START:
            if transaction.received_start:
                raise DuplicateStartError()
                
        elif packet.type == PacketType.DATA:
            if not transaction.received_start:
                self.buffer_orphaned_packet(packet)
                raise MissingStartError()
                
        elif packet.type == PacketType.END:
            if not transaction.is_complete():
                raise IncompleteTransferError()
                
    def progressive_checksum(self, packet):
        # Maintain running checksum as packets arrive
        transaction = self.active_transactions[packet.transaction_id]
        transaction.update_checksum(packet.data)
```

Implementation Notes:

1. Progressive Validation:
- Validate packets as they arrive rather than waiting for completion
- Maintain running checksums
- Track byte ranges to identify gaps immediately

2. Transaction Management:
- Keep track of active transactions
- Clean up completed/cancelled transactions
- Handle timeout scenarios

3. Error Recovery:
- Implement explicit negative acknowledgments (NACKs)
- Request retransmission of specific byte ranges
- Handle out-of-order packet arrival

4. Configuration Options:
```python
class FileTransferConfig:
    def __init__(self):
        self.max_active_transactions = 100
        self.transaction_timeout = 300  # seconds
        self.retry_attempts = 3
        self.chunk_size = 1024  # bytes
```

These mitigations would:
1. Prevent incorrect file reassembly
2. Enable detection of missing START packets
3. Support parallel transfers
4. Improve error detection and recovery
5. Maintain backward compatibility with existing systems

closed 2025-12-19 by thomas-bc