nasa/fprime #2224: PolyDb updates to enforce single writer

gcgandhi, 2023-08-23:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  PolyDb |
---
## Feature Description

Enforcing a single writer to measurement to prevent accidentally writing to the incorrect measurement .  This was brought up by Adit.

## Rationale

There is nothing preventing a component from accidentally writing a value to the incorrect measurement.

matt392code, 2025-01-11:
[image]

matt392code, 2025-01-11:
Proposed files for implementing the single writer enforcement in PolyDb:
[polydb-files.txt](https://github.com/user-attachments/files/18383826/polydb-files.txt)
Let me explain the key aspects of each file:

1. **PolyDb.fpp**:
- Defines the component interface with guarded ports for setValue and getValue
- Adds new registerWriter port and command
- Defines events for unauthorized writes and successful registrations

2. **PolyDbCfg.fpp**:
- Sets maximum number of entries
- Defines invalid writer ID constant

3. **PolyDb.hpp/.cpp**:
- Implements writer registration and validation
- Uses mutex for thread safety
- Maintains map of writer entries with registration status
- Implements command handlers and port handlers

4. **Tester.hpp/.cpp**:
- Sets up unit test infrastructure
- Provides methods for testing writer registration
- Includes port connection setup

5. **CMakeLists.txt**:
- Configures build system
- Sets up dependencies
- Registers unit tests

Key Features:
- Thread-safe access using mutex
- Clear error reporting for unauthorized writes
- Command interface for writer registration
- Proper event logging
- Compatibility with existing PolyDb usage

LeStarch, 2025-02-26:
@gcgandhi what would you recommend as an implementation here?  There (currently) is no indication of the sender within the context of PolyDb.

We theoretically could add the object ID to the port and have the autocoder add it in the call. This would give us the information to check.  It isn't safe against a malicious component, but would work against accidental sets.

Let me know so we can evaluate this proposal.

LeStarch, 2025-02-26:
@gcgandhi how would you recommend we implement this?  There is no way to determine the caller within the context of `PolyDb`. Thus we could only add an extra argument for the calling developer to set.  However, this is no less error-prone than just the measurement index.

The only way I see is to add a caller ID in via the autocode, but that would affect every port in the system.  Thoughts?

gcgandhi, 2025-02-27:
@LeStarch I have not thought through a full solution, but I can say that for a similar type of problem we were able to constraint the potential writers to assigned Address Space via MMU protections.  I can't offer any ideas beyond what you proposed without much more thought/analysis, as you already covered what I would have thought of.