nasa/fprime #2225: add license option when generating project scaffold with `fprime-util new --project`

LeStarch, 2023-08-23:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Option to add an [OSI](https://opensource.org/about) approved license when generating project scaffold with `fprime-util new --project` - perhaps with default to Apache 2.0 to match core fprime codebase.

## Rationale

As to allow for ease of integration and contributions as an fprime community project

thomas-bc, 2023-08-30:
Requested by the community: see https://github.com/fprime-community/fprime-tools/issues/155

thomas-bc, 2023-08-30:
This feels to me like one of those things that sits on the edge of whether this is the responsibility of the framework to support. My gut says no. It's very easy to add on the user/github side, most likely safer, and probably out of the scope of developing FSW/F´. 

I am tempted to vote in favor of closing this issue as "Won't implement".

@LeStarch thoughts?

capsulecorplab, 2023-09-02:
Isn't the entire point of `fprime-tools` to encourage best practices? To omit a license option would suggest a license is more of an afterthought.

thomas-bc, 2023-12-07:
Not planned for the time being. Although we can reconsider if more users in the community are requesting this.

closed 2023-12-07 by thomas-bc