nasa/fprime #2226: Remove unnecessary parsers arguments when applicable

LeStarch, 2023-08-23:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Parsers are architected in a way where they all inherit from a top level parser that defines a bunch of stuff which is supposedly shared. Only, some of this stuff if not used in some commands, bloating the help text with unnecessary and wrong information, because those options don't actually do anything.
E.g.: `fprime-util format --help` will print those options
```
  -d DEPLOY, --deploy DEPLOY
                        F prime deployment directory to use. May contain multiple build directories.
  -p PATH, --path PATH  F prime directory to operate on. Default: cwd, /Users/chammard/Work/fp/fprime-tools.
  --build-cache BUILD_CACHE
                        Overrides the build cache with a specific directory
  -v, --verbose         Turn on verbose output.
  --ut                  Run command against unit testing build type
```

We should find a way to remove those extra parsing options when they are not needed. Somehow, it isn't as straightforward as you'd think to remove an argument from a parser. **This may be a hint that our parser architecture is not adapted to the growing functionality of `fprime-util`**, which doesn't only revolve around builds anymore.

## Rationale

Un-bloat help texts.

LeStarch, 2024-03-27:
@thomas-bc....is this necessary?  If not, could we close?

thomas-bc, 2024-03-27:
@LeStarch I'd like to keep this open. It's not the top of my priority list, but it would most definitely be nice to have.

The fact that this happens in the first place also sounds to me like the one-parser-fits-all architecture that we currently have may not be the most optimal fit for the growing functionality of `fprime-util` - which has expanded past its original build helper scope.

matt392code, 2025-01-11:
Proposed implementation of a modular argument system for F Prime:

Proposed lightweight solution for removing unnecessary parser arguments:

1. Create argument groups that map to specific command needs:
```python
BUILD_ARGS = {
    "deploy": ("-d", "--deploy", "F prime deployment directory to use"),
    "build_cache": ("--build-cache", "Overrides the build cache"),
    "ut": ("--ut", "Run command against unit testing build type")
}

DEV_ARGS = {
    "format_style": ("--style", "Style configuration for format command"),
    "check_all": ("--all", "Run all checks")
}

COMMON_ARGS = {
    "verbose": ("-v", "--verbose", "Turn on verbose output")
}
```

2. Modify command registration to only add relevant arg groups:
```python
def register_format_command(cls):
    parser = cls.get_parser()
    # Only add common args + specific needed args
    add_arg_group(parser, COMMON_ARGS)
    add_arg_group(parser, DEV_ARGS)
    # Explicitly avoid BUILD_ARGS for format command
```

This approach:
- Requires minimal architecture changes
- Keeps existing parser inheritance 
- Commands explicitly choose which arg groups they need
- Removes help text bloat
- Can be implemented incrementally per-command

This solution provides a concrete solution without requiring major refactoring, while still addressing the core issue of unnecessary arguments appearing in help text.