nasa/fprime #2231: Dictionary versions with just a Git Commit SHA produce a confusing error message

LeStarch, 2023-08-23:
From @nbering

| | |
|:---|:---|
|**_F´ Version_**| 3.1.0 |
|**_Affected Component_**| Ref Application |
---
## Problem Description

GDS suggests upgrading when builds from a Git checkout - without tags - produce a dictionary with just a commit SHA as version, like this:

```xml
<dictionary topology="Ref" framework_version="7cb9feed0" project_version="7cb9feed0">
```

```
[ERROR] Dictionary version  is not in supported range: 0.0.0-9.9.9. Please upgrade fprime-gds.
```

## How to Reproduce

1. Build the Ref app from an F´ checkout that doesn't have Git Tags in the local history
2. Run `fprime-gds` as documented in the [Install guide](https://nasa.github.io/fprime/INSTALL.html#checking-your-f-installation)

I have a longer-form description of the problem and how I encountered it [in this blog post](https://nicholasbering.ca/fprime/2022/08/10/fprime-git-tags/).

## Expected Behaviour

Suggested solutions:
- An error about version format instead of version range, possibly suggesting that release tags are missing
- Run with a warning instead of an error when the version number can't be evaluated
- Fix `fprime-util` so it doesn't produce invalid versions

LeStarch, 2023-08-23:
From @ThibFrgsGmz: I have it too! I thought it was due to my wobbly personal setup at the moment. 😅

LeStarch, 2023-08-23:
@nbering: 

> I have it too! I thought it was due to my wobbly personal setup at the moment. 😅

Thanks for the "me, too!" comment. I was worried I might be reporting something only I encountered because I did something unexpected. There's a bunch of reasons people might not end up with tags in their local checkout, though.

LeStarch, 2023-08-23:
@nbering:

I was just looking through the project to get a better sense of whether I could write a fix for this. Looks like there's a comment that defines the expected behavior as ignoring version strings that are not formatted with `major.minor.patch`.

https://github.com/fprime-community/fprime-gds/blob/48efee990481e3c152c1b0fb5a50821a07a1ca55/src/fprime_gds/common/loaders/xml_loader.py#L120-L121

LeStarch, 2023-08-23:
From @thomas-bc:

I just had this happen to me, good thing I had seen this issue here before!

Yes it looks we're catching that there's an issue with the value of the version, but not really handling it (https://github.com/fprime-community/fprime-gds/blob/bbae3d6275b18706dc757a965fce8a1a4214adfd/src/fprime_gds/common/loaders/xml_loader.py#L128-L129). `dict_version` becomes the empty tuple, which raises the DictionaryException. We should instead detect that we're in this special case, and probably just raise a warning.
I'll get to it when I have time, but feel free to open a PR if you want to!

evanmayer, 2023-10-16:
I think have this issue too. At least, I have the same error message.

Also, because it might be related, I'd like to make sure when I'm developing that I'm running the latest version of F' in my project's `fprime` directory. Is there a recommended way to update it?

LeStarch, 2023-11-29:
Revisit after fpp-to-dictionary

closed 2023-12-14 by LeStarch