nasa/fprime #2234: Add A Verify F´  Tools Version at Install

LeStarch, 2023-08-23:
Users following tutorials get confused when the prompts do not match the tool output.  We should add a step to verify tool versions.

benfrstr, 2024-02-15:
Could I work on this?

thomas-bc, 2024-02-15:
@benfrstr feel free to make any contributions to the tutorial repositories you think would make user experience / understanding better! 
For this specific issue, I think it might not be necessary anymore, since we reference the install guide and have a better process to keep tutorials in sync than we had at the time this ticket was created.

benfrstr, 2024-02-15:
@thomas-bc I don't want to contribute anything unnecessary, so it's up to you. I will look for another good first issue if this issue is no longer applicable.

thomas-bc, 2024-02-16:
Closing as the tutorials have been modified to reference the install guide.

closed 2024-02-16 by thomas-bc