nasa/fprime #2253: Broken Documentation link

timcanham, 2023-08-24:
| | |
|:---|:---|
|**_F´ Version_**|3.2.x |
|**_Affected Component_**|  none |
---
## Problem Description

Navigating through the docs produces a broken link:

https://nasa.github.io/fprime/v3.3.0/Design/general.md

A description of the problem with sufficient detail to understand the issue.

## How to Reproduce

1. Navigate to https://nasa.github.io/fprime/v3.3.0/
2. Click on `Design` link.
3. Page not found!

## Expected Behavior

A working link

thomas-bc, 2023-08-25:
I think our autodocs script does not generate the `Design` section for the versioned documentation snapshots. I'm not sure why (or if that's the desired behavior), but all other snapshots (see [here](https://nasa.github.io/fprime/)) do **not** contain the Design section. 
We should either remove that broken link in in [docs/v3.3.0](https://nasa.github.io/fprime/v3.3.0/), or generate it if desired.

LeStarch, 2024-04-01:
Removed the broken link.

closed 2024-04-01 by LeStarch