nasa/fprime #2273: GDS won't pass an array as a value for a command

neel-puri, 2023-09-13:
| | |
|:---|:---|
|**_F´ Version_**| 3.3.2 |
|**_Affected Component_**|  fprime-gds |
---
## Problem Description

If a custom array type is defined (i.e. an array of 2 F32's) the gds won't accept the input. If a command is defined such that it only has one input, the array of 2 F32's, then the system is capable of building everything and starting the gds. When attempting to send that defined command, after pressing "send command" the gds simply displays "unknown error" under the "mnemonic" drop down menu. 

In the console of the web-browser developer tools, an error is sent with the message "Failed to validate all arguments" without any further context or information.

There are no issues with any other commands. 

## How to Reproduce

1. Create a custom data type (the one I did was `arrType = [4] F32`)
2. In a new component define an asynchronous command that takes the `arrType` as a value (what I did was `async command TEST_COMMAND(Vals: arrType @< just a test of the type)`
3. The command handler can be the simple auto coded one. It doesn't need to do anything (this was mine: 
`void <component> :: TEST_COMMAND_cmdHandler( const FwOpcodeType opCode, const U32 cmdSeq, Components::arrType vals ) {
    // Release Event about data being sent to process
    this->log_ACTIVITY_LO_JUST_ANNOUNCE_AN_EVENT();
    // TODO
    this->cmdResponse_out(opCode,cmdSeq,Fw::CmdResponse::OK);
  }`
4. Start the `fprime-gds` in the normal way
5. Click on the `<component>.TEST_COMMAND` option in the mnemonic drop down
6. Fill out the fields and send the command
7. The described error should appear

## Expected Behavior

The expected behavior is announces the given event. The event does nothing other than post an event to the event's tab that says "it worked." 

In the long term sense, it should provide the function with the data provided in the command array as a series of F32s so that the function can manipulate it at will.

SMorettini, 2023-09-17:
@neel-puri  can you check if my pull request https://github.com/fprime-community/fprime-gds/pull/141 solves your problem?

neel-puri, 2023-09-18:
@SMorettini I have two questions. 1) how do I test your pull request? 2) If the answer to the first question is I re-install the fprime-tools from scratch, then the new changes did not fix the issue. 

So the issue persists, but I have a feeling I am not properly testing the pull request though.

thomas-bc, 2023-09-21:
Hey @neel-puri, the process to test a PR would be to clone the code and install locally. So in this case, since it's in SMorettini's repo, you would do the following:

```
git clone https://github.com/SMorettini/fprime-gds
cd fprime-gds
git checkout Fix-scalar-input-forms
pip install fprime-gds
```

Then go back to your project (no need to rebuild since this is a change to the GDS only), and re-run the `fprime-gds` command.
I'll also be taking a look at this

thomas-bc, 2023-09-25:
You may want to double check that you're actually using the version of the GDS you just installed. Did you run `fprime-gds` in the same shell session? I tested myself and it seemed to work fine.

neel-puri, 2023-09-25:
Ok yes. That change fixed the issue. The mistake I made was that I didn't do "pip install ." within the virtual environment. A very silly mistake in hindsight, but oh well I guess. 

Thank you so much @thomas-bc. The change in that branch fixes the issue. Should I close the issue? or should I wait for the change to migrate to devel branch?

thomas-bc, 2023-09-25:
We'll take care of closing the issue once the fix is merged in! Thanks for confirming @neel-puri

closed 2023-10-30 by thomas-bc