nasa/fprime #2282: GDS command argument type-checking edge cases

thomas-bc, 2023-09-22:
| | |
|:---|:---|
|**_F´ Version_**| 3.3.2 |
|**_Affected Component_**|  GDS |
---
## Problem Description

There is an issue when type-checking an argument in the GDS (see [here](https://github.com/fprime-community/fprime-gds/blob/0604b543ccfbb0faa2c5560c6598e4e54572a3a5/src/fprime_gds/flask/static/addons/commanding/arguments.js#L242-L251))
If the argument of your command is a enum or struct, type.name will carry the value of the C++ type, which will be ModuleName::EnumName. Now say the name of the module is Foo, that gets you Foo::MyEnum, so the argument.type.name[0] == 'F' condition will be true. So this kind of restricts the usage of module names that start with U, I or F.

## How to Reproduce

1. Create a module/deployment with a name that starts with an F, I or U
2. Create a custom type (say an enum) in said module, and a command that uses that type as an argument
3. Sending a command will fail to validate the argument

## Expected Behavior

Should validate the argument successfully.

## Proposed solution

Abstract away the logic of type checking in a function, which will actually check whether the type name matches U8, U16, U32 etc...

thomas-bc, 2023-09-22:
This was revealed by https://github.com/fprime-community/fprime-gds/pull/141

LeStarch, 2023-11-29:
Fixed!

closed 2023-11-29 by LeStarch