nasa/fprime #2284: Multi-field Command Validation

Tetragramm, 2023-09-25:
| | |
|:---|:---|
|**Version**| 3.3.2 |
|**Component**| GDS |
---
## Problem Description

When sending a command with the ground station, validation always returns the cursor to the first field, even if the error is in a subsequent field.  For example, when first filling out the fields, after every number typed, you must click on the field again to type the next digit.

## How to Reproduce

1. Create a command with a structure containing at least 3 values.
2. Click on the second field and type a number.
3. Note that cursor is placed in the first field.
4. Clear all fields
5. Type a number in first field.
6. Type a number in the second field.
7. Note that cursor is placed in the first field again.
[image]

## Expected Behavior

The validation should not move the cursor from the field it is in.  It should not matter which field you're typing in first, and it absolutely shouldn't require N clicks to type an N-digit number.

As requested in: https://github.com/fprime-community/fprime-gds/issues/143

LeStarch, 2023-10-12:
Related to #2312

LeStarch, 2023-10-13:
Should be fixed by: https://github.com/fprime-community/fprime-gds/pull/147

closed 2023-10-25 by thomas-bc