nasa/fprime #2285: ld warnings when building unit tests

bocchino, 2023-09-25:
**Issue:** Since updating to the latest Mac OS tools and updating to the latest fprime/devel, I'm seeing warnings like this when building unit tests on Mac OS:
```
ld: warning: ignoring duplicate libraries: '../../../lib/Darwin/libFw_Com.a', '../../../lib/Darwin/libFw_Types.a'
```
Everything seems to build and run fine, and the unit tests pass, but we should investigate these warnings to find the root cause.

**How to reproduce:** Run `fprime-util build --ut` in any build module that has unit tests. The warning occurs at the step where the build system links the unit test executable.

LeStarch, 2023-09-25:
The cause is these libraries are being specified on the link-line twice.  Where this is legal in a Linux environment (single pass linker), this is a warning in macOS (multi-pass linker).

Going to poke around a bit more and see if I can determine the cause...but I don't know how easy it will be to solve.

LeStarch, 2023-09-26:
I don't think this can be fixed. The issue stems from the fact that the UT depends on libraries that one of its dependents also depends on.  Rather than de-duplicating and reordering the list, CMake is just linking the same library twice (which is correct behavior on single-pass linkers).

closed 2023-09-26 by LeStarch