nasa/fprime #2295: Documentation Link Broken

timcanham, 2023-10-04:
| | |
|:---|:---|
|**_F´ Version_**| 3.x |
|**_Affected Component_**| None  |
---
## Problem Description

Broken documentation link in GDS Test API docs:

[image]

## How to Reproduce

1. Go to https://nasa.github.io/fprime/UsersGuide/dev/testAPI/user_guide.html 
2. Click on link in "An example script to run the Ref App deployment without a GDS Tool can be found [here](https://nasa.github.io/Ref/scripts/run_ref_for_int_test.sh)."
3. 404

## Expected Behavior

Valid page

thomas-bc, 2024-01-04:
This is likely out-dated with the introduction of the pytest fixture. Refactoring of the docs is to be addressed by https://github.com/nasa/fprime/issues/2295

LeStarch, 2024-03-28:
Should be fixed by #1902

closed 2024-03-28 by LeStarch