nasa/fprime #2297: ByteStreamDriverModel Undefined Build Error With Latest Tutorial

thvarkey, 2023-10-05:
### Discussed in https://github.com/nasa/fprime/discussions/2290

<div type='discussions-op-text'>

<sup>Originally posted by **astroesteban** October  2, 2023</sup>
Hi y'all,

I was going through the [Hello World: Integration and Testing With F´ Deployments](https://fprime-community.github.io/fprime-tutorial-hello-world/docs/Deployments.html) page just to get acquainted with the latest F Prime updates (the updates to prime-util are phenomenal by the way) and I keep getting this issue when trying to build the `HelloWorldDeployment`:

```cpp
[ 97%] Generating HelloWorldDeploymentTopologyAppAi.xml, Ports_RateGroupsEnumAi.xml
fpp-to-xml
/workspace/MyProject/HelloWorldDeployment/Top/instances.fpp:108.27
  instance comDriver: Drv.ByteStreamDriverModel base id 0x4000 \
                          ^
error: undefined symbol ByteStreamDriverModel
gmake[3]: *** [HelloWorldDeployment/Top/CMakeFiles/HelloWorldDeployment_Top.dir/build.make:113: HelloWorldDeployment/Top/HelloWorldDeploymentTopologyAppAi.xml] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:8300: HelloWorldDeployment/Top/CMakeFiles/HelloWorldDeployment_Top.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:8224: HelloWorldDeployment/CMakeFiles/HelloWorldDeployment.dir/rule] Error 2
gmake: *** [Makefile:2418: HelloWorldDeployment] Error 2
[ERROR] CMake erred with return code 2
```

I'm using the latest F Prime version on `devel` (commit f34d596cf41b8fea594a3313b972c027fc28a3de) and running the following commands:

```sh
cd MyProject/HelloWorldDeployment
fprime-util build -j $(nproc)
```

I also opted for the default responses when creating the component and deployment unless otherwise instructed by the tutorial.

Any thoughts as to what could be going wrong?</div>

LeStarch, 2023-10-05:
Note: `fprime-util new` is intended to be used with the latest tag of F´.  That is why this issue occured (as devel was updated).

We fixed this issue here: https://github.com/fprime-community/fprime-tools/pull/167 and that fix will be consistent with F´ at the next release!

closed 2023-10-05 by LeStarch

thvarkey, 2023-10-05:
Packet XML parsing error: Channel helloWorld.GreetingCount does not exist

I am following the same tutorial and encountered the same issue regarding ByteStreamDriveModel. However, now I am seeing the XML error stated above. 

The tutorial mentions this below. Is there an update required in the HelloWorldDeployment/Top/HelloWorldDeploymentPackets.xml
    <ignore>
        ...
        <channel name="helloWorld.GreetingCount"/>
    </ignore>

thomas-bc, 2023-10-05:
@thvarkey did you make sure to update the HelloWorldDeploymentPackets.xml, as shown [here](https://github.com/fprime-community/fprime-tutorial-hello-world/blob/fb7cc713ce98a2c7a60cd4086f0abb79d74016c6/HelloWorldDeployment/Top/HelloWorldDeploymentPackets.xml#L81) ?

thvarkey, 2023-10-05:
@thomas-bc Correct, I updated the ignore section as you mentioned in the link. Question: Do I need to go back to previous section of the tutorial to regenerate any of the files?

thvarkey, 2023-10-05:
@thomas-bc Please find some of the details:

Packet SystemRes2 size 101/111
Packetizing SystemRes3 (7)
Packet SystemRes3 size 85/111
Packet XML parsing error: Channel helloWorld.GreetingCount does not exist
make[3]: *** [HelloWorldDeployment/Top/CMakeFiles/HelloWorldDeployment_Top.dir/build.make:177: HelloWorldDeployment/Top/HelloWorldDeploymentPacketsAc.cpp] Error 255
make[2]: *** [CMakeFiles/Makefile2:8841: HelloWorldDeployment/Top/CMakeFiles/HelloWorldDeployment_Top.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:8705: HelloWorldDeployment/CMakeFiles/HelloWorldDeployment.dir/rule] Error 2
make: *** [Makefile:2413: HelloWorldDeployment] Error 2

thomas-bc, 2023-10-05:
Hmm no you shouldn't need to... but maybe just for the sake of being sure, try
```
fprime-util purge -f
```
Then re-generate and build:
```
fprime-util generate
fprime-util buid
```

thvarkey, 2023-10-05:
@thomas-bc Unfortunately, I am still getting the same error. Please see below the ignored packets section from my HelloWorldDeploymentPackets.xml

    

    <ignore>
        <channel name="cmdDisp.CommandErrors"/>

        <channel name="helloWorld.GreetingCount"/>
    </ignore>

LeStarch, 2023-10-05:
`Packet XML parsing error: Channel helloWorld.GreetingCount does not exist` means the HelloWorld component does not yet define the telemetry channel `GreetingCount`.

Make sure that step was followed.

thvarkey, 2023-10-05:
@LeStarch @thomas-bc That was the error. Thank you so much