nasa/fprime #2298: Use static allocation in OSAL

sobkulir, 2023-10-08:
| | |
|:---|:---|
|**_F´ Version_**| 3.3.2 |
|**_Affected Component_**|  OSAL |
---
## Feature Description

Right now, resource-owning OSAL APIs typically have a `POINTER_CAST m_handle` class member to hold a handle to the underlying OS resource. The resource is typically expected to be dynamically allocated in an initialization method, or in the case of Mutex in the constructor.

Therefore, I'd suggest:
1) Enforce static allocation across OSAL
2) Do not expect the Mutex to allocate from the heap in the constructor (!!). This maybe should be filed as a bug.

## Rationale

Well, dynamic allocation in safety-critical software is a topic in itself, but from what I could understand F Prime tries to enforce doing dynamic allocation only in the initialization phase. However, some of the OSAL types "silently" allocate from the heap during the operational phase, e.g., when opening a File in FileDownlink component. There are many more examples. This may not be of big concern when a lot of RAM is available, but for deployments with a few hundred kilobytes of RAM this issue becomes relevant.

## Suggested implementation

I thought of _templating_ out the handle member, e.g.:
```c++
template <typename T>
class Mutex {
    public:
        void lock(); //!<  lock the mutex
        void unLock(); //!<  unlock the mutex

    private:
        T m_handle; //!<  Stored handle to mutex
};
``` 

If templates are to be avoided, another option would be using _macros_, as in the case of `TaskIdRepr`:
```c++
#if defined(TGT_OS_TYPE_VXWORKS) || (FW_BAREMETAL_SCHEDULER == 1)
    typedef int TaskIdRepr;
#elif defined(TGT_OS_TYPE_LINUX) || defined(TGT_OS_TYPE_DARWIN)
    typedef pthread_t TaskIdRepr;
#endif

class TaskId {
  public:
      TaskId();
      ~TaskId();
      bool operator==(const TaskId& T) const;
      bool operator!=(const TaskId& T) const;
      TaskIdRepr getRepr() const;
  private:
      TaskIdRepr id;
};
```

## More details
The affected APIs are: Mutex, Queue, File, Directory, Task, InterruptLock, and WatchdogTimer. In case of the Queue, its size would need to either be determined at compile time or the API would need to allow passing an external buffer.

timcanham, 2023-10-09:
@sobkulir To keep the OSAL layer portable, we made the choice to pass handles this way since some platforms we are targeting (like Linux) only allow this method. The current OSAL layer does not preclude static allocation behind the interface if the platform or homegrown OSAL implementation supports it. For instance, one of our internal projects is implementing a basic file system where the file name is used to look up an entry in a static table, so there is no dynamic memory allocation but the API is preserved.

sobkulir, 2023-10-10:
@timcanham Thank you for your response!

> The current OSAL layer does not preclude static allocation behind the interface if the platform or homegrown OSAL implementation supports it.

Maybe I'm missing something, but I think this is not correct. For example, since you mentioned file system, here's our implementation of `File::open()` in Zephyr. It's quite hacky because we used the `NATIVE_INT_TYPE m_fd` meant for file descriptor to hold a pointer. 

```c++
File::Status File::open(const char* fileName, File::Mode mode, bool include_excl) {
    struct fs_file_t *file = reinterpret_cast<struct fs_file_t *>(k_malloc(sizeof(struct fs_file_t)));

    if (file == nullptr) {
        FW_ASSERT(0);
        return NO_SPACE;
    }

    fs_mode_t flags = modeToZephyrFlags(mode);

    fs_file_t_init(file);
    int rc;
    
    rc = fs_open(file, fileName, flags);
    if (rc < 0) {
        DEBUG_PRINT("Error opening file %s: %d\n", fileName, rc);
        this->m_lastError = rc;
        k_free(file);
        return NOT_OPENED;
    }

    // Dangerous!
    this->m_fd = reinterpret_cast<NATIVE_INT_TYPE>(file);
    this->m_mode = mode;
    return OP_OK;
}
```

The full implementation can be found here: [sobkulir/fprime-zephyr-app/fprime-zephyr/Os/File.cpp](https://github.com/sobkulir/fprime-zephyr-app/blob/main/fprime-zephyr/Os/File.cpp#L25). I'd be happy if you have an idea on how to avoid the malloc.

I totally see your point on avoiding the templates and macros as much as possible. At the same time, I would like to find a compromise because on some OSes this clashes with the rule of `Avoid dynamic allocation after initialization`, as shown above, or the `Directory::open()` example from the first comment.

sobkulir, 2023-10-10:
@timcanham I would use this API, but allocate the `struct fs_file_t handle` on stack or globally. I'm not 100% sure, and this is what we still need to check internally, but the subsequent `fs_open()` doesn't use any dynamic memory and there's a compilation-time parameter in Zephyr on how many files can be opened at the same time. The kernel preallocates the I/O buffers statically.

timcanham, 2023-10-10:
@sobkulir So your objection is not so much the open call, but the `k_malloc` call (which I missed, sorry!).

sobkulir, 2023-10-10:
@timcanham Exactly! It's not a big issue, like the handle is a few bytes. Same for threads, mutex, etc. My main point is that I feel like the API is actively going against the `Avoid dynamic allocation after initialization` principle.

sobkulir, 2023-10-10:
Haha I was just typing a comment suggesting that -- you mean put it in `PlatformTypes.h`, right?

timcanham, 2023-10-10:
Or that too!

sobkulir, 2023-10-10:
Nice! Either would resolve this as far as I can tell. Thanks for the discussion! I don't have the time right now, but in about a month I would be happy to implement this. Do you have a preference over `PlatformTypes.h` or `FpConfig.h`? Personally, I feel like `PlatformTypes.h` fits this use case more, as `FpConfig.h` is project-specific, while the issue here is platform-specific.

timcanham, 2023-10-10:
Good idea! I opened this discussion: https://github.com/nasa/fprime/discussions/2306

timcanham, 2023-10-10:
Let's move the trade about where to define it to that discussion.

LeStarch, 2024-05-10:
I am going to close this issue.  The format used for Os::File and Os::Task seems to work, and as such this is just part of the refactor steps.

closed 2024-05-10 by LeStarch