nasa/fprime #2302: Make GoogleTest and STest as  Submodules

LeStarch, 2023-10-09:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Google test should be made into a submodule. Since users should just include F´ then google test should be shipped as a submodule.  This would prevent the "google test downloads during generate" problem.

thomas-bc, 2023-10-09:
When this is done, we will likely need to add a command to checkout submodules recursively in `fprime-util new --project`, [here](https://github.com/fprime-community/fprime-tools/blob/devel/src/fprime/cookiecutter_templates/cookiecutter-fprime-project/hooks/post_gen_project.py#L22-L46)

LeStarch, 2023-10-09:
@thomas-bc is there harm doing it now?  Seems like it would be future-proofing.

thomas-bc, 2023-11-16:
GTest refactoring is now tracked https://github.com/nasa/fprime/issues/1998

closed 2023-11-29 by thomas-bc