nasa/fprime #2326: Seg Fault While Running Unit Test

Steph41, 2023-10-18:
| | |
|:---|:---|
|**_F´ Version_**| 3.3.3|
|**_Affected Component_**|  Unit Testing |
---
## Problem Description

A description of the problem with sufficient detail to understand the issue.

Ran into a seg fault while attempting to run the Unit Tests of the LED Blinker workshop. 

Output when running prime-unit check 
> 0% tests passed, 1 tests failed out of 1
> 
> Total Test time (real) =   0.29 sec
> 
> The following tests FAILED:
> 	  1 - Components_Led_ut_exe (SEGFAULT)

Output from lldb

> (other-venv) MT-205418:Led scontrer$ lldb ../../build-fprime-automatic-native-ut/bin/Darwin/Components_Led_ut_exe 
> (lldb) target create "../../build-fprime-automatic-native-ut/bin/Darwin/Components_Led_ut_exe"
> Current executable set to '/Users/scontrer/workshop/build-fprime-automatic-native-ut/bin/Darwin/Components_Led_ut_exe' (x86_64).
> (lldb) run
> Process 25141 launched: '/Users/scontrer/workshop/build-fprime-automatic-native-ut/bin/Darwin/Components_Led_ut_exe' (x86_64)
> Process 25141 stopped
> * thread # 1, stop reason = EXC_BAD_ACCESS (code=1, address=0x7ff800000000)
>     frame #0: 0x0000000100eb696c dyld
> ->  0x100eb696c: movl   (%rax), %edx
>     0x100eb696e: andl   $-0x2, %edx
>     0x100eb6971: cmpl   $0xfeedface, %edx         ; imm = 0xFEEDFACE 
>     0x100eb6977: jne    0x100eb6930
> Target 0: (Components_Led_ut_exe) stopped.
> (lldb) bt
> * thread # 1, stop reason = EXC_BAD_ACCESS (code=1, address=0x7ff800000000)

Micheal Starch said it's due to an incompatibility in my GCC version. He commented out the sanitizers in fprime and it compiled correctly 
>(other-venv) MT-205418:Led scontrer$ gcc --version
>Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
> Apple clang version 12.0.0 (clang-1200.0.32.28)
> Target: x86_64-apple-darwin22.6.0
> Thread model: posix
> InstalledDir: /Library/Developer/CommandLineTools/usr/bin

LeStarch, 2023-10-19:
We've seen this before where the sanitizer libraries linked in are not compatible with the clang compiler versions.  This causes a segfault before `main()` is even run.  Previously the fix was to upgrade the compilers, but in this case it was not possible.

Joshua-Anderson, 2023-10-20:
I had the same issue, See #1964 for more details. Clang 12 long deprecated, but macos doesn't always seem to upgrade xcode tools when upgrading, so you have to manually trigger an xcode tools upgrade.

LeStarch, 2024-03-28:
I am going to close this, as the solution would be to turn off the sanitizers

closed 2024-03-28 by LeStarch