nasa/fprime #2327: Unit tests fail because gcov can't merge coverage analysis

bocchino, 2023-10-18:
**How to reproduce it:** On Mac OS, run `fprime-util check` in a component directory, then revise the test code and rerun. On OS X, gcov emits lots of warnings like `... gcda: cannot merge previous run count: corrupt object tag ...`. Sometimes the tests run successfully with these warnings. Otherwise there are test failures, including segmentation faults.

**Notes:**

* I've observed this behavior on Mac OS. I can't confirm whether this behavior occurs on other systems.
* When I run `fprime-util check --coverage`, the output says `[INFO] Scrubbing leftover .gcda files` and I don't see this behavior. When I run `fprime-util check` without `--coverage` I don't see the info message and I do see the warnings.

LeStarch, 2023-10-19:
While `fprime-util` does remove `.gcda` files, it only does so when the `--coverage` flag is supplied.  However, the run of the unit test (`check` w/ or w/o `--coverage`) produces these files. Thus in half the cases the error still appears. This is because `check` is a direct build target, and the `check --coverage` target is a composite of: clean, check, and gcovr.

To fix this, we need to make `check` a composite target that runs the clean and check.  `check --coverage` should then be a composite target of  this new check and the `gcovr` run. This will result in the same behavior for `check --coverage` while fixing `check`

LeStarch, 2023-11-29:
Fixed!

closed 2023-11-29 by LeStarch