nasa/fprime #2330: Overriding Baremetal Os Abstraction

ethancheez, 2023-10-19:
| | |
|:---|:---|
|**_F´ Version_**| 3.3.2 |
|**_Affected Component_**|  Os/Baremetal/ |
---
## Problem Description

The [`fprime/Os/Baremetal/CMakeLists.txt`](https://github.com/nasa/fprime/blob/devel/Os/CMakeLists.txt#L84) file always adds the Baremetal Os abstractions of File, FileSystem, IntervalTimer, Mutex, Queue, SystemResources, Task when defining the `FPRIME_USE_BAREMETAL_SCHEDULER` flag. So even if an external package that attempts to create a specific Os abstraction (i.e. File), the `Os/Baremetal/___.cpp` versions will always be used instead of the external package.

## Reproduce Bug

We can use the existing `IntervalTimer.cpp` file in `fprime-arduino` as an example, as that is *expected* to override the `Os/Baremetal/IntervalTimer.cpp` within `fprime`.

1. Simply call the `IntervalTimer` `getRawTime()` function. It will call the function inside of the `fprime` `Os/Baremetal` version instead of `fprime-arduino`'s version.

## Suggestion
The `FPRIME_USE_BAREMETAL_SCHEDULER` flag should only replace the Task Scheduler to use the Baremetal implementation, not add all of the other Baremetal Os abstractions. The baremetal package(s) should be the ones adding those baremetal abstractions (see how `fprime-arduino` is doing it [here](https://github.com/fprime-community/fprime-arduino/blob/devel/Arduino/Os/CMakeLists.txt#L26).

## Why
A `File.cpp` abstraction for the `fprime-arduino` package is currently in development as it is needed for [`fprime-artemis-cubesat`](https://github.com/fprime-community/fprime-artemis-cubesat) to replace the Baremetal implementation, which currently does nothing. 

## Temporary Solution
The following lines were commented out in the [CMake](https://github.com/nasa/fprime/blob/devel/Os/CMakeLists.txt#L84) file:
```
# list(APPEND SOURCE_FILES "${CMAKE_CURRENT_LIST_DIR}/Baremetal/File.cpp")
# list(APPEND SOURCE_FILES "${CMAKE_CURRENT_LIST_DIR}/Baremetal/FileSystem.cpp")
# list(APPEND SOURCE_FILES "${CMAKE_CURRENT_LIST_DIR}/Baremetal/IntervalTimer.cpp")
# list(APPEND SOURCE_FILES "${CMAKE_CURRENT_LIST_DIR}/Baremetal/Mutex.cpp")
# list(APPEND SOURCE_FILES "${CMAKE_CURRENT_LIST_DIR}/Baremetal/Queue.cpp")
# list(APPEND SOURCE_FILES "${CMAKE_CURRENT_LIST_DIR}/Baremetal/SystemResources.cpp")
# list(APPEND SOURCE_FILES "${CMAKE_CURRENT_LIST_DIR}/Baremetal/Task.cpp")
```
Instead, these are linked within the Os abstraction of `fprime-arduino` [here](https://github.com/fprime-community/fprime-arduino/blob/devel/Arduino/Os/CMakeLists.txt#L26).

For example, if I choose to override the Baremetal `File.cpp` version, I simply replace:
```
"${FPRIME_FRAMEWORK_PATH}/Os/Baremetal/File.cpp"
``` 
with:
```
"${CMAKE_CURRENT_LIST_DIR}/File.cpp"
```

LeStarch, 2023-10-19:
Agreed.  This is one of the many drivers behind #1685.  The intent there is to allow selection of implementations for each thing in `Os`

LeStarch, 2024-11-22:
I think this is fixed.

closed 2024-11-22 by LeStarch