nasa/fprime #2336: Why the test api of gds has double the histories?

LeStarch, 2023-10-23:
### Discussed in https://github.com/nasa/fprime/discussions/2153

<div type='discussions-op-text'>

<sup>Originally posted by **SMorettini** July 24, 2023</sup>
In [api.py](https://github.com/fprime-community/fprime-gds/blob/928a264420a1f2b566c4f1a88e7fceae861fa619/src/fprime_gds/common/testing_fw/api.py#L41C1-L44C65) there are the following lines:
```
# these are owned by the GDS and will not be modified by the test API.
self.aggregate_command_history = pipeline.histories.commands
self.aggregate_telemetry_history = pipeline.histories.channels
self.aggregate_event_history = pipeline.histories.events
```
Why do we need to create the history objects for GDS inside the integration tests? From my understanding, the pipeline of GDS and the pipeline for the integration tests are two different pipelines and both contain their history objects. Could someone clarify this point?

I notice that I run my integration tests for hours and the ram usage keeps increasing because I was clearing the test histories but the histories indicated in the code above were not cleaned by GDS. 

P.s.: In my setup, GDS is started separately, am I don't it wrong? </div>

SMorettini, 2023-10-24:
From my understanding what happens is:
* The GDS process has a pipeline containing the 3 histories(commands, events, channels). These are cleaned up as soon as the data reaches the GUI.
* api.py contains:
    * a pipeline with 3 histories(commands, events, channels). Never cleanup.
    * three histories for the test itself. Cleanup at the beginning of each test in `start_test_case`

The histories that are never cleaned are also never used. So my temporary fix was to add in my fork of `fprime-gds`
```
pipeline.coders.remove_command_consumer(self.aggregate_command_history)
pipeline.coders.remove_channel_consumer(self.aggregate_telemetry_history)
pipeline.coders.remove_event_consumer(self.aggregate_event_history)
```
Is this the solution we want to go for or do you think there are better solutions?

LeStarch, 2023-10-24:
There is a flag to allow you to specify no GUI. In this case the whole UI pipeline should be disabled leaving just the data movement framework but not the UI rendering stack.

```
fprime-gds -g none
```

However, it seems that flag is still starting Flask, so that is a bug to correct

LeStarch, 2023-10-24:
We should also verify in the TestAPI stack that only the test histories are used and we don't get a redundant copy.

LeStarch, 2023-10-24:
Duplicate of #2337

closed 2023-10-24 by LeStarch