nasa/fprime #2337: GDS can consume the full system memory if the GUI is not use

SMorettini, 2023-10-24:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**| fprime-gds   |
---
## Problem Description

The history inside `fprime-gds` is cleaned only when the data are sent to the GUI. If for some reason the GUI is not opened(for example when using the option `-g none`) the history is never cleaned. If Fprime and gds is run for enough time they will eventually take all the memory available and crash. 

## How to Reproduce

1. Start fprime-gds with the option `-g none`
2. Start an Fprime executable(it would be better if the executable publish channel with a lot of data and/or very frequently)
3. Let it run for a while(like half an hour) and check the memory consumption of your device.

## Expected Behavior

Memory consumption should not grow without limits.

## Possible fixes

A fix I thought but I'm not sure it's acceptable is to add a maximum history size in `SelfCleaningRamHistory`. When the history reaches the maximum size, the first element added in the history can be removed avoiding unlimited growth.

I was also thinking of disabling the history when there is no GUI but from the structure of GDS I'm not sure is possible.

LeStarch, 2023-10-24:
This is further discussed here: https://github.com/nasa/fprime/issues/2336

In short, `-g none` should not start the flask/UI app at all, yet it is.

LeStarch, 2023-10-24:
Fixed by:  https://github.com/fprime-community/fprime-gds/pull/149

LeStarch, 2023-10-25:
Fix was merged.

closed 2023-10-25 by thomas-bc