nasa/fprime #2350: Several errors in GitHub Pages

ConstStar, 2023-10-30:
https://nasa.github.io/fprime/Tutorials/CrossCompilationSetup/CrossCompilationSetupTutorial.html
Under the title of ‘3. Installing the Toolchain’
```
# For users running on 32-bit ARM
curl -Ls https://developer.arm.com/-/media/Files/downloads/gnu-a/10.2-2020.11/binrel/gcc-arm-10.2-2020.11-x86_64-arm-none-linux-gnueabihf.tar.xz | tar -JC /opt/toolchains --strip-components=1 -x
```
There are no arm-linux-gnueabi-gcc files in gcc-arm-10.2-2020.11-x86_64-arm-none-linux-gnueabihf.tar.xz.
It should be arm-none-linux-gnueabihf-gcc For 32-bit ARM hardware

https://fprime-community.github.io/fprime-tutorial-hello-world/docs/Deployments.html
Under the title of ‘Adding The Hello World Component’
Path MyDeploymment/Top/instances.fpp should be changed to HelloWorldDeployment/Top/instances.fpp

Thanks!

geoffreylgv, 2024-01-07:
Hi @ConstStar @thomas-bc, does this issue active? I cannot understand what is needed to do, can you elaborate more.
I went through and I see that nothing is needed, things are up-to-date

thomas-bc, 2024-01-08:
Seems the errors have been corrected, thanks for pointing it out!

closed 2024-01-08 by thomas-bc

geoffreylgv, 2024-01-08:
My pleasure