nasa/fprime #2372: TlmPacketizer: Allows users to individually turn on and off groups of packets

Joshua-Anderson, 2023-11-15:
| | |
|:---|:---|
|**_Affected Component_**| TlmPacketizer  |
---
## Feature Description

Users currently set an integer packet level in TlmPacketizer, and all telemetry packets with a level less than or equal to the current level will be sent.

Current Command Interface:
```
SET_LEVEL 3
# Packets with level 1, 2, and 3 will be sent
```

This proposal recommends removing packet levels in favor of packet groups. 

Allow users to enable and disable telemetry packet groups on a group by group basis. Allow users to specify in the packet list xml file if the packet group should default to enabled or disabled. Ideally packet group would be an enum rather than an integer, allowing users to name packet groups, but using the packet list xml file to generate a FPP enum may be challenging

Proposed Command Interface:
```
ENABLE_GROUP 4
# Packets of group 4 will now be sent
ENABLE_ALL
# Send all telemetry groups
DISABLE_GROUP 1
# Packets of group 4 will no longer be sent
DISABLE_ALL
# Stop transmission of all vehicle telemetry
```

## Rationale

This allows spacecraft operators to carefully tune what telemetry is sent based on the operational mode, providing more flexibility than merely increasing the overall volume of telemetry sent. One example of a use case for this capability would be only sending rover mobility telemetry while actively driving.

Joshua-Anderson, 2023-11-21:
This is a subset of issue #2210, which recommends the addition of more telemetry management capabilities to F'

timcanham, 2025-02-04:
@bocchino Perhaps if the FPP isn't set in stone yet, we could change the packet spec from "level" to "group" in anticipation of this.

closed 2025-12-19 by thomas-bc