nasa/fprime #2397: Fix native-image builds

thomas-bc, 2023-11-29:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

FPP, FPL etc... fails to build - investigate
https://github.com/fprime-community/native-images-action/tree/main

thomas-bc, 2023-12-05:
One of the latest successful run building fpp native images https://github.com/fprime-community/fpp/actions/runs/6459740810/job/17540152759 includes this warning

        ********************************************************************************
        The arguments ['--plat-name=manylinux_2_28_x86_64'] were given via `--global-option`.
        Please use `--build-option` instead,
        `--global-option` is reserved for flags like `--verbose` or `--quiet`.

        This deprecation is overdue, please update your project and remove deprecated
        calls to avoid build errors in the future.
        ********************************************************************************

Latest runs now fail with 
```
error: option --plat-name not recognized
```

Ticket on pip : https://github.com/pypa/pip/issues/11859

LeStarch, 2023-12-07:
https://github.com/fprime-community/native-images-action/releases/tag/v0.2.3 this fixes all the problems.